ros-navigation/navigation2 #1110: Can we remove this use_dwa parameter?

yathartha3, 2019-09-11:
Can we remove this use_dwa parameter? Not part of this PR, but if so, can we file an issue and do it soon? It seems silly to keep something that was deprecated in ROS.

_Originally posted by @mkhansen-intel in https://github.com/ros-planning/navigation2/pull/1045_

SteveMacenski, 2019-09-11:
Look at what the parameter does- its useful but perhaps it should be renamed

mkhansen-intel, 2019-09-12:
Can you provide the link to the lines of code again? I seem to remember it gives a warning saying it's deprecated. If it was already deprecated in ROS, we shouldn't have it at all in ROS2 in my opinion

SteveMacenski, 2019-09-12:
https://github.com/ros-planning/navigation2/blob/e342010bdbfd9ab69f285d19759a42341cda3fca/nav2_dwb_controller/dwb_plugins/src/standard_traj_generator.cpp#L91

and 

https://github.com/ros-planning/navigation2/blob/e342010bdbfd9ab69f285d19759a42341cda3fca/nav2_dwb_controller/dwb_plugins/src/limited_accel_generator.cpp#L67

The exception says the parameter is invalid (which circling back is what this ticket is about) but the functionality it represents is still useful (I think? I haven't looked at these generators in any real detail but I dont think David would have included it for fun if it wasn't useful to someone). Its just that the parameter is no longer the appropriate way to enable it.

I kind of like that this exception is thrown as a reminder that you can get DWA-like behavior from DWB if you choose but I can gt behind that the parameter can be removed and there's potentially a better way to handle it. I suppose I mixed in my head the `use_dwa` parameter and the functionality it represents. We can refactor to remove the parameter but still make it clear that generator is to enable DWA like behavior.

Or delete the other generator all together if you don't think it adds value.

crdelsey, 2019-09-13:
The parameter existed in ROS 1 because base local planner had DWA and Trajectory Rollout built in, and the parameter was the way to choose between them.

In DWB, you choose between the two options by choosing which trajectory generator to load. Each trajectory generator throws an exception if the parameter is inconsistent, but it is just there for backward compatibility.

I believe the way it works is limited_accel_generator is DWA and standard_traj_generator is Trajectory Rollout. So, limited_.. throws an exception if use_dwa is false, and standard_.. throws an exception if it is true.

SteveMacenski, 2019-09-17:
Yeah. 

Lets go ahead and remove the parameter, but document that if you want DWA-like behavior, to swap out the generator plugin to the other

mkhansen-intel, 2019-10-25:
Note from my original issue that there are tests that are checking this param and will need to be updated also.

SteveMacenski, 2019-11-21:
Anyone taking ownership of this?

PR #1394 (2019-12-11): Remove the obsolete parameter "use_dwa"

---

## Basic Info

| Info | Please fill out this column |
| ------ | ----------- |
| Ticket(s) this addresses   | fixes #1110 |
| Primary OS tested on | Ubuntu |
| Robotic platform tested on | TB3 in Gazebo|

---

## Description of contribution in a few bullet points

* Removes the use_dwa parameter
closed 2019-12-11 by crdelsey