ros-navigation/navigation2 #1124: NavigateToPose action failing with CycloneDDS

Jconn, 2019-09-16:
## Bug report

**Required Info:**

- Operating System:
Ubuntu 18.04

- Version or commit hash:
bd0fe8bf238c69bc41ea29680a7a37bd3fb60022 (tag: 0.2.4)
- DDS implementation:
  - cyclone dds 

#### Steps to reproduce issue

- clone cyclone dds into your ros2 dashing workspace
- build ros2 dashing from source
- build navigation2 
- launch navigation2 stack and rviz with 

```
ros2 run rviz2 rviz2 -d ~/nav2_ws/src/navigation2/navigation2/nav2_bringup/launch/nav2_default_view.rviz
ros2 launch nav2_bringup nav2_bringup_launch.py use_sim_time:=False autostart:=False map:=~/empty_map.yaml
```

- set a NavigateToPose goal through rviz

#### Expected behavior
robot navigates to pose, as with other dds implementation
#### Actual behavior

rviz and nav2 throw exceptions 
from nav2
```
[bt_navigator-8] [INFO] [bt_navigator]: Begin navigating from current location to (-1.81, 0.10)
[bt_navigator-8] [WARN] [rcl.logging_rosout]: Publisher already registered for provided node name. If this is due to multiple nodes with the same name then all logs for that logger name will go out over the existing publisher. As soon as any node with that name is destructed it will unregister the publisher, preventing any further logs for that name from being published on the rosout topic.
[bt_navigator-8] terminate called after throwing an instance of 'rclcpp::exceptions::RCLError'
[bt_navigator-8]   what():  error taking goal response: error not set
[ERROR] [bt_navigator-8]: process has died [pid 18362, exit code -6, cmd '/home/mekaworks/nav2_ws/install/nav2_bt_navigator/lib/nav2_bt_navigator/bt_navigator __node:=bt_navigator __params:=/tmp/tmpttfsvmyw'].
```

from rviz
```
[INFO] [rviz2]: Setting goal: Frame:map, Position(-0.855215, 0.571564, 0), Orientation(0, 0, 0.539519, 0.841974) = Angle: 1.13973
terminate called after throwing an instance of 'rclcpp::exceptions::RCLError'
  what():  error taking status: error not set

```

I have a few other nodes successfully running with cyclone dds using built-in and custom message types

I have had a lot of trouble with fastRTPS crashing and opensplice being slow. My first experience with cyclone + comments on https://github.com/ros-planning/navigation2/issues/1094 have got me interested in it.

mkhansen-intel, 2019-09-17:
@rotu has tested CycloneDDS with dashing, see his blog post here:  https://discourse.ros.org/t/blog-post-2-ros-2-dashing-with-navigation/10656

Your error might be that you are running with `autostart:=False`. If you're running that way you have to use the `Startup` button in Rviz to activate the nodes, or you can do it via command line tools but that's much slower. 

Have you used the `Startup` button in this case? If so, did you see the lifecycle_manager node print: `[lifecycle_manager-11] [INFO] [lifecycle_manager]: The system is active`

If those are OK, and the system is active, can you please attach the entire log file?

Also, did you try with FastRTPS and do you get the same issue?

Jconn, 2019-09-17:
The stack is working great for me with fastrtps and opensplice.

I hadn't seen rotu's blog post, thanks for pointing me to that.

I have tried building ros2 dashing from source with cyclone in that ws, that is unsuccesful.

I have also tried building cyclone using ros2 installed from debian pkgs. That is working for some of my packages but not for nav2. My other packages use custom msgs but not custom actions.

I just now tried not building cyclone and using the released debian package, that yields the same output as building cyclone does. 

I haven't switched over to the testing repos for ros2 debian packages, but I guess I should try that now. 

This probably isn't a bug with the nav2 stack, I probably just have some work to do with cyclone config. I'm comfortable closing this at your discretion.  

The log output for my NavigateToPose crash is here:
[nav_log.txt](https://github.com/ros-planning/navigation2/files/3623797/nav_log.txt)

rotu, 2019-09-18:
Hmmm... I definitely had some [mysterious RMW issues with ROS 2 Navigation built from source](https://github.com/ros-planning/navigation2/issues/1056) that I had difficulty reproducing. I suspect switching to the binary distribution of Nav2 will fix your issue.

The other thing to look at is [this gigantic backport](https://github.com/ros-planning/navigation2/commit/ec6ae7cf4e7a804f51776d8877ad65bdfc94f2d5), which was not included in [my bringup](https://www.youtube.com/watch?v=TqJsTDli0zU). It may be benign, but the `--ros-args` argument was backported even though it is not supported in Dashing.

Jconn, 2019-09-18:
I just tested with 0.2.3 built from source and confirmed the issue still reproduces. Also tested debian 0.2.3 and the issue reproduces. The last big difference I see between your success and my failure is in my dashing debian packages coming from the release repo and not the test repo.

My setup for getting other packages to play nice with cyclone when building it from source was definitely suspect. For instance, the only way I could get two nodes using cyclone to pub/sub to eachother was by building from source was sourcing my cyclone_ws before sourcing /opt/ros/dashing. Installing cyclone got rid of all that nonsense.

rotu, 2019-09-18:
@jconn, so what’s currently built from source? rmw_cyclonedds_cpp? rmw_implementation? navigation2?

And what does the navigation log look like?

SteveMacenski, 2019-09-18:
> was by sourcing my cyclone_ws before sourcing /opt/ros/dashing. Installing cyclone got rid of all that nonsense.

That makes sense to me - things weren't getting connected because the packages needing to be overlayed hadn't been yet

rotu, 2019-09-18:
A weird thing I see in your nav_log.txt are some huge numbers associated with the point cloud data (though it could just be someone logging durations in nanoseconds instead of seconds)

> [world_model-4] [INFO] [global_costmap.global_costmap]: Created an observation buffer for source obj_pc, topic /object_detection/obj_pc, global frame: map, expected update rate: 0.00, observation persistence: 1000000000.00
> ...
> [dwb_controller-5] [INFO] [local_costmap.local_costmap]: Created an observation buffer for source pointcloud, topic /camera/depth/color/points, global frame: map, expected update rate: 3000000000.00, observation persistence: 0.00

SteveMacenski, 2019-09-18:
Pretty sure that's just a nanosecond issue.

Jconn, 2019-09-18:
I tried the following setups:
**setup 1**
built from source:
- navigation2 version 0.2.3 / 0.2.4
- rmw_cyclonedds_cpp
- ros2-dashing

**setup 2**
installed through debian packages:
- navigation2 version 0.2.3
- rmw_cyclonedds_cpp
- ros2-dashing

**setup 3** 
built from source:
- navigation2 version 0.2.4

installed through debian packages:
- rmw_cyclonedds_cpp
- ros2-dashing

When using cyclone, all of these setups were able to pub/sub between navigation2 and other nodes, and crashed when I sent a NavigateToPose action to the nav stack. I have a pointcloud2 source that is updating the costmap, and I can visually confirm that stream is working.

It's only when I send the NavigateToPose action through rviz or through my own python code that nav2 spits out the following error:

```
[bt_navigator-8] [INFO] [bt_navigator]: Begin navigating from current location to (-0.23, -1.31)
[bt_navigator-8] terminate called after throwing an instance of 'rclcpp::exceptions::RCLError'
[bt_navigator-8]   what():  error taking goal response: error not set
[ERROR] [bt_navigator-8]: process has died [pid 6552, exit code -6, cmd '/home/mekaworks/nav2_ws/install/nav2_bt_navigator/lib/nav2_bt_navigator/bt_navigator __node:=bt_navigator __params:=/tmp/tmpnn8it7_9'].
```

Those large numbers on the pointcloud data were to work around this [behavior](https://github.com/ros-planning/navigation2/issues/1054) in 0.2.3, thank you for reminding me about that that!

rotu, 2019-09-18:
I am flummoxed. The only other thing I can think of is that I'm using a [simpler behavior tree](https://github.com/RoverRobotics/openrover-ros2/blob/09a8218984e460efdcccf6c46402ddd0a0931bbf/openrover_demo/config/navigate.xml), whereas you're using [navigate_w_replanning_and_recovery.xml](https://github.com/ros-planning/navigation2/blob/44dcd6054ce2922dc9481a75ac07871450612af4/nav2_bt_navigator/behavior_trees/navigate_w_replanning_and_recovery.xml). Something ilke the `ClearEntireCostmap` BT Node might be causing trouble.

mkhansen-intel, 2019-09-18:
@Jconn - have you tried running the action [tutorials](https://github.com/ros2/demos/tree/master/action_tutorials) using cyclone? That would help us understand if it's a cyclone actions issue or a nav2 issue. Although TBH I haven't run those myself.

Jconn, 2019-09-18:
I hadn't tried any of the action tutorials or examples, that's a good idea.

I just got through running the following examples: 
examples_rclpy_minimal_action_client/server
examples_rclcpp_minimal_action_client/server
run action_tutorials fibonacci_action_client/server

and they all worked with ros2-dashing and rmw-cyclone installed through debian packages

rotu, 2019-09-20:
@jconn, I just tried rerunning my stack since the 0.2.4 sync, and I'm not seeing the crash you are.
Is your underlay built from source or installed from apt? Have you deleted your overlay build folder and rebuilt?

Jconn, 2019-09-20:
Is my underlay the ros2 packages that compose the ros2 dashing release? Those are installed from apt. 

I remove my build folder every time I try a different approach to getting cyclone to work, just in case

rotu, 2019-09-21:
Yes, that’s exactly what I mean - the stuff like rmw_implementation, rclcpp, and rosidl.

Btw I confirmed that those clear costmap recovery bt nodes don’t cause a crash on my machine.

For now, I’m out of ideas :-(

Jconn, 2019-09-21:
that's okay, It's probably just an environment error on my side. 

Thanks for spending the cycles walking through this with me. It's encouraging that cyclone should be working and does work for others. I'm sure I'll stumble onto the fix sooner or later and I'll comment when I do.

I'll close this for now, because I'm skeptical that the problem is in this software package.

airuchen, 2019-10-01:
Dear @Jconn 
This issue happens to me currently. I'm curious if you have any further progress.

```
[bt_navigator-7] terminate called after throwing an instance of 'rclcpp::exceptions::RCLError'
[bt_navigator-7]   what():  error taking goal response: error not set
[ERROR] [bt_navigator-7]: process has died [pid 6854, exit code -6, cmd '/opt/ros/dashing/lib/nav2_bt_navigator/bt_navigator __node:=bt_navigator __params:=/tmp/tmpkna63_u6'].
```

Jconn, 2019-10-01:
Nope, still busted for me. Every few days I come back to this and try something new

rotu, 2019-10-01:
@airuchen please clip a little more context from the logs. What else does bt_naviator log? Is this always preceded by a warning about a node name collision?

airuchen, 2019-10-01:
@Jconn  Thanks.
@rotu It works fine when I set estimate, but breaks when setting goal. Here are the logs.
```
[bt_navigator-7] [INFO] [bt_navigator]: Begin navigating from current location to (0.73, 0.26)
[bt_navigator-7] [WARN] [rcl.logging_rosout]: Publisher already registered for provided node name. If this is due to multiple nodes with the same name then all logs for that logger name will go out over the existing publisher. As soon as any node with that name is destructed it will unregister the publisher, preventing any further logs for that name from being published on the rosout topic.
[bt_navigator-7] terminate called after throwing an instance of 'rclcpp::exceptions::RCLError'
[bt_navigator-7]   what():  error taking goal response: error not set
[ERROR] [bt_navigator-7]: process has died [pid 18033, exit code -6, cmd '/opt/ros/dashing/lib/nav2_bt_navigator/bt_navigator __node:=bt_navigator __params:=/tmp/tmpkwem41qs'].
```

And these are what rviz shows
```
terminate called after throwing an instance of 'rclcpp::exceptions::RCLError'
  what():  error taking status: error not set
```

rotu, 2019-10-01:
Estimates are processed by the amcl node. Goals are processed by the bt_navigator node.

Try removing the node_name argument to bt_navigator in your launch file. If your launch file uses the LifecycleNode action (which requires the node_name argument), switch to the Node action (which doesn’t require it). While node name collisions *can* be harmless, I suspect this one might not be.

airuchen, 2019-10-01:
Thx for helping. I took out the line, and the warning disappeared. Still, the error remains.
```
[bt_navigator-7] [INFO] [bt_navigator]: Begin navigating from current location to (0.76, 0.10)
[bt_navigator-7] terminate called after throwing an instance of 'rclcpp::exceptions::RCLError'
[bt_navigator-7]   what():  error taking goal response: error not set
[ERROR] [bt_navigator-7]: process has died [pid 23657, exit code -6, cmd '/opt/ros/dashing/lib/nav2_bt_navigator/bt_navigator __params:=/tmp/tmp_u74imgs'].
```
I will try on another device to check if it's just an env issue.

crdelsey, 2019-10-01:
@Jconn Sounds like you are not the only one experiencing the problem. I'm going to reopen the issue till we fix it or figure out where to file an issue upstream

orduno, 2019-10-08:
I'm running the debugger and am down to this [method](https://github.com/ros2/rmw_cyclonedds/blob/1205ab5ea223a72253db9c36b19707978b122a9e/rmw_cyclonedds_cpp/src/rmw_node.cpp#L1863) where is not "taking" the response request.

We should probably file an issue at `rmw_cyclonedds`.

orduno, 2019-10-09:
We've found the source of the problem. Here's some background:

`BTNavigator` is an action server to `NavigateToPose` but also creates a goal subscription:

https://github.com/ros-planning/navigation2/blob/6bfa8c48ea0291fff540eb7aead6e5e511130bd7/nav2_bt_navigator/src/bt_navigator.cpp#L65

On which it's callback uses an action client to send the action to itself:

https://github.com/ros-planning/navigation2/blob/6bfa8c48ea0291fff540eb7aead6e5e511130bd7/nav2_bt_navigator/src/bt_navigator.cpp#L247

The action server and client are created on different nodes and are spinning independently. However, for some reason, some action msgs are getting mixed up, i.e. when the action is received through the action server (not triggering the goal subscription callback and therefore not calling action client), I'm seeing some action messages coming in on node with the action client. :thinking: 

I've filed an issue upstream https://github.com/ros2/rmw_cyclonedds/issues/47. While a bit unorthodox, it's still a valid use case of actions and is not a problem on FastRTPS.

For now, if you're using the action interface to sent the goal, you can remove the internal action client, `self_client`, and that works.

Jconn, 2019-10-10:
I got rid of `self_client` and now cyclonedds is working great. Thanks so much for root causing this @orduno . I'm comfortable closing this now, or we can leave it open until rmw_cyclonedds is fixed

SteveMacenski, 2019-10-17:
Closing from discussion above and active PR in the rmw

closed 2019-10-17 by SteveMacenski