ros-navigation/navigation2 #1126: Compilation Error: No member named ‘register_param_change_callback’

lbegani, 2019-09-17:
Error log below while building nav2 master with ros2 master, perhaps introduced by this - https://github.com/ros2/rclcpp/commit/89119c6422a8788d985f038b4841ce4f2edcf0ed

```
Starting >>> nav2_rviz_plugins
--- stderr: nav2_dynamic_params                                                                                        
/home/lkumarbe/nav2_master_ws/src/navigation2/nav2_dynamic_params/src/dynamic_params_validator.cpp: In constructor ‘nav2_dynamic_params::DynamicParamsValidator::DynamicParamsValidator(rclcpp::Node::SharedPtr, bool)’:
/home/lkumarbe/nav2_master_ws/src/navigation2/nav2_dynamic_params/src/dynamic_params_validator.cpp:29:10: error: ‘using element_type = class rclcpp::Node {aka class rclcpp::Node}’ has no member named ‘register_param_change_callback’; did you mean ‘set_on_parameters_set_callback’?
   node_->register_param_change_callback(
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
          set_on_parameters_set_callback
make[2]: *** [CMakeFiles/nav2_dynamic_params.dir/src/dynamic_params_validator.cpp.o] Error 1
make[1]: *** [CMakeFiles/nav2_dynamic_params.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make: *** [all] Error 2
```

## Bug report

**Required Info:**

- Operating System: Ubuntu 18.04
  - 
- Version or commit hash: 4973026a4397ffe5a56a47b2fb32c5c25aa8daa6
  - 
- DDS implementation: Fast-RTPS
  - 

#### Steps to reproduce issue

```
Build nav2 master with ROS2 master
```

bpwilcox, 2019-09-17:
This `nav2_dynamic_params` package is deprecated anyway, so we should perhaps just remove it, as was suggested in #933

SteveMacenski, 2019-09-17:
I opened a PR to do it. https://github.com/ros-planning/navigation2/pull/1128

22 days ago I think I implied that I was going to do it but didn't realize I implied that. So I'll do that now

ruffsl, 2019-09-17:
Resolved: https://github.com/ros-planning/navigation2/pull/1128

closed 2019-09-17 by ruffsl