ros-navigation/navigation2 #1134: Document steps for running Nav2 with Cartographer

mkhansen-intel, 2019-09-18:
## Feature request
PR #1057 added support for running Nav2 with cartographer, we need to update the documentation with a short "How to"

SteveMacenski, 2019-09-18:
Both for future proofing and such, I don't think the word "Cartographer" should appear, I think it should be a document/tutorial focused on using SLAM with the stack (or outside localizer) where the ending example is cartographer potentially

mkhansen-intel, 2019-09-18:
I changed the name of the issue. However, to be clear, the work we've done has only been with Cartographer, so the documentation will show that. I don't have the time or need to test it with other SLAM implementations, if someone in the community wanted to do that, they could add to or change the documentation via a follow-on PR. The intent right now is to document what we have tested so others can use it.

SteveMacenski, 2019-09-18:
Well the point is to not make it seem like Cartographer <> Navigation are a pairing, its really any SLAM implementation, where cartographer is just the easiest one to point at right now.

Any SLAM system that properly broadcasts a map and TF will be compliant. Explaining the QoS settings and topics --> running is really all I would think a document like that would include, what else is there to it?

closed 2019-09-25 by mlherd