ros-navigation/navigation2 #1135: Update System test to use refactored launch files in nav2_bringup

mkhansen-intel, 2019-09-18:
# Feature request
The system tests are using custom / duplicated launch files. We should refactor them to use the newly refactored launch files in nav2_bringup. This will have 2 benefits:
1. Test the nav2_bringup_launch.py file itself
2. Reduce launch file code duplication

#### Feature description

The main test_system_launch.py should include the nav2_bringup_launch.py for launching the nodes and only add the additional components such as gazebo and static_transform_publisher.

#### Implementation considerations

Ideally, I think the system test should also use the parameters file in the nav2_bringup package. See #813

yathartha3, 2019-09-19:
@mkhansen-intel I saw this after I submitted PR #1140. Now that I found this issue, it seems like there is a bigger change that needs to be done.

mkhansen-intel, 2019-09-23:
Yes, this is a bigger refactor, regardless of the change you already made

orduno, 2019-10-21:
@mkhansen-intel I think #1243 addressed this, can we close the issue?

mkhansen-intel, 2019-10-25:
Yes, I think #1243 did this, thanks!

closed 2019-10-25 by mkhansen-intel