ros-navigation/navigation2 #1145: Navigation2 build uses behaviortree headers from underlay

rotu, 2019-09-20:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.05
- Version or commit hash:
  - bd0fe8bf238c69bc41ea29680a7a37bd3fb60022
  - ros-dashing-behaviortree-cpp	2.5.2-1bionic.20190918.161712
- DDS implementation:
  - N/A

#### Steps to reproduce issue
1. Install Dashing and have BehaviorTree.CPP in your underlay. Make sure you have a clang warning
2. Install BehaviorTree.CPP 8969bbe204aec8540fb2efe095ae482d0104199c from source in your overlay.
3. Set your compiler to clang
4. Build your workspace

#### Expected behavior
The build succeeds.

#### Actual behavior
The build fails, since nav2 picks up behavior tree headers from the underlay instead of the overlay:

```
$ colcon build --packages-select nav2_behavior_tree
Starting >>> nav2_behavior_tree
--- stderr: nav2_behavior_tree                             
In file included from /home/dan/ros2_ws/src/navigation2/nav2_behavior_tree/src/behavior_tree_engine.cpp:15:
In file included from /home/dan/ros2_ws/src/navigation2/nav2_behavior_tree/include/nav2_behavior_tree/behavior_tree_engine.hpp:23:
/opt/ros/dashing/include/behaviortree_cpp/bt_factory.h:199:17: error: lambda capture 'this' is not used [-Werror,-Wunused-lambda-capture]
        return [this](const std::string& name, const NodeParameters& params)
```

rotu, 2019-09-20:
@dirk-thomas, do you see anything in https://github.com/ros-planning/navigation2/blob/bd0fe8bf238c69bc41ea29680a7a37bd3fb60022/nav2_behavior_tree/CMakeLists.txt or https://github.com/BehaviorTree/BehaviorTree.CPP/blob/ros2-version-2.x/CMakeLists.txt that would cause the former to pick up headers in the latter from the underlay instead of the overlay?

dirk-thomas, 2019-09-20:
The former package uses `ament_target_dependencies` which makes sure that the include directories are ordered by the workspace order (see https://github.com/ament/ament_cmake/blob/99d316aab1d8652950d67afa2356baa04f51a207/ament_cmake_target_dependencies/cmake/ament_target_dependencies.cmake#L61). So that seems fine.

In the second package I don't see where `ament_INCLUDE_DIRS` is being set (so I would suspect it to be empty hence irrelevant). Besides that I don't see anything wrong.

I assumed this is a fresh build of the workspace or at least with the CMake cache files deleted?

rotu, 2019-09-20:
No, it's not a fresh build. I'm unable to reproduce it reliably right now, so I think some cache files may be persisting.

dirk-thomas, 2019-09-20:
If you built the workspace before without the package being in the overlay CMake had already called the location of the package in the underlay and won't search again. So you certainly want to use `--cmake-clean-cache` once after adding the package to the overlay.

rotu, 2019-09-20:
Thanks, @dirk-thomas. That's almost certainly it. It was probably due to using `colcon build --continue-on-error`.

For my own future reference, it's this line in `~/ros2_ws/build/nav2_behavior_tree/CMakeCache.txt`
```
//The directory containing a CMake configuration file for behaviortree_cpp.
behaviortree_cpp_DIR:PATH=/opt/ros/dashing/share/behaviortree_cpp/cmake
```

closed 2019-09-20 by rotu