ros-navigation/navigation2 #1151: Nav2 dashing now uses a different default navigation goal topic name than RViz

rotu, 2019-09-20:
#1096 changed the `bt_navigator` to listen for navigation goals on the "/goal_pose" topic instead of the "/move_base_simple/goal" topic.

https://github.com/ros-planning/navigation2/blob/ec6ae7cf4e7a804f51776d8877ad65bdfc94f2d5/nav2_bt_navigator/src/bt_navigator.cpp#L58-L61

The corresponding change in RViz has not been backported:
https://github.com/ros2/rviz/pull/455

I don't know whether this was an appropriate change to backport. The fix is to either backport this change on RViz or to undo the change in Navigation2.

## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.05
- Version or commit hash:
  - bd0fe8bf238c69bc41ea29680a7a37bd3fb60022
- DDS implementation:
  - CycloneDDS

#### Steps to reproduce issue
Launch nav2 and rviz and specify a goal pose.

#### Expected behavior
bt_navigator responds to the goal pose

#### Actual behavior
bt_navigator never acknowledges the message

SteveMacenski, 2019-09-20:
Its not a bug, if you're running bleed of both, it should be aligned. I made that change. RVIZ bleed now publishes at `goal_pose` too

rotu, 2019-09-20:
@SteveMacenski  Please reread the issue and reopen. I assume by "bleed" you mean Eloquent. I'm talking about Dashing, which has recently backported this breaking change:

https://github.com/ros-planning/navigation2/blob/dashing-devel/nav2_bt_navigator/src/bt_navigator.cpp#L58-L61

https://github.com/SteveMacenski/rviz/blob/dashing/rviz_common/default.rviz#L66-L67

SteveMacenski, 2019-09-20:
ah, @crdelsey can we get that updated in the next sync? 

Its going to be a few weeks at the earliest before the package manager will reflect that.

rotu, 2019-09-20:
To be clear, are you un-backporting the change in Nav2 Dashing or backporting the change in RViz Dashing?

SteveMacenski, 2019-09-20:
un-backporting the change in Nav2 Dashing. I don't have control over RViz and that's not a change that should have been backported in either

SteveMacenski, 2019-09-20:
I'll fix this, but also just having a config for your rviz that remaps the topic is a 15 second thing. I'd recommend from just a company perspective to not be using the default rviz configs because you never know who's going to do what to it when.

closed 2019-09-20 by SteveMacenski

rotu, 2019-09-21:
@SteveMacenski Thank you. I wasn't using the RViz default. I was using [my own rviz file](https://github.com/RoverRobotics/openrover-ros2/blob/master/openrover_demo/config/default.rviz) and wondering why Nav2 was broken :-)

rotu, 2019-10-03:
@SteveMacenski Did this get released? Why don't I see this change in `ros-dashing-nav2-bt-navigator	0.2.4-1bionic.20191001.071451`?

crdelsey, 2019-10-07:
This will end up in the 0.2.5 binaries.