ros-navigation/navigation2 #1152: AMCL is not "Adaptive"

yathartha3, 2019-09-20:
**Adaptive** Monte Carlo Localization will benefit from using an adaptive sample set size. The algorithm that is currently implemented is the "**Augmented** Monte Carlo Localization".

**Why this matters:**
Adaptively changing the number of sample allows for computational efficiency.
Example, when you start up the robot, you might need 10000 particles. However, once it has localized, it probably needs less particles to localize.
The algorithm is in page 264 of Probabilistic Robotics book, and is named `KLD_Sampling_MCL`.

For some reason the legacy AMCL code has this commented out saying
>   // Can't (easily) combine low-variance sampler with KLD adaptive
  // sampling, so we'll take the more traditional route.

So at this point, the **A** in AMCL implementation stands for **augmented**, and not **adaptive** (which was surprising to me).

yathartha3, 2019-09-20:
I can implement the adaptive version. However, I wanted someone to take a look to confirm this issue, and how we might implement the adaptive version.

SteveMacenski, 2019-09-21:
I would definitely not disagree that there are benefits from being derived from actually doing this sampling, but before starting that since it should be relatively straight forward, I would try to create a set of metrics and environments to make sure that those changes reflect substantive improvements across many use-cases since this is a (very) package alot of people rely on with their current set of parameters.  

I almost wonder if you're motivated and there's a business case if it might be better to just redo AMCL entirely based on modern tools. Someone in C++land must have a heavily used pf implementation, same with the math and random sampling from a distribution. Add in the plugin based motion models for omni/diff/ackermann and that's off to the races.

SteveMacenski, 2019-09-21:
Repeating back to you: That comment also indicate that someone tried to do KDL sampling but the variance in particles were consistently low enough that something wasn't either working right or worth the complexity. I dont know if that gives you a clue as to why it was removed, but I'd have to pull out a notebook and my copy of Dieter's book to figure it out

yathartha3, 2019-09-21:
> I would definitely not disagree that there are benefits from being derived from actually doing this sampling, but before starting that since it should be relatively straight forward, I would try to create a set of metrics and environments to make sure that those changes reflect substantive improvements across many use-cases since this is a (very) package alot of people rely on with their current set of parameters. Metrics like improved localization / response, amount of additional computation time, which then would affect the maximum number of particles a particular microprocessor could have at peak, etc
> 

Since many people have their current set of params, the 'augmented' can be the default. With the adaptive version, there won't be introduction of new params, since there are already all the necessary params, but are not being used. But how it will affect other things, is something I don't know.

> I almost wonder if you're motivated and there's a business case if it might be better to just redo AMCL entirely based on modern tools. Someone in C++land must have a heavily used pf implementation, [...]

Plugins for different motion models is a really good idea.

I am not saying AMCL has to be made adaptive, but rather just pointing it out since I was surprised it was not already doing it.

yathartha3, 2019-10-01:
Might be useful when we are testing on bigger maps, or with compute limited computers. However, users can always try to tune the number of particles using params. Will close for now. Reopen if needed.

closed 2019-10-01 by yathartha3