ros-navigation/navigation2 #1154: dwb oscillation critic not respecting some params

Jconn, 2019-09-22:
## Bug report

**Required Info:**

- Operating System:
Ubuntu 18.04 
- Version or commit hash:
navigation 0.2.4
- DDS implementation:
opensplice

#### Steps to reproduce issue

set oscillation parameters in the config yaml file 

    #default is .2
    Oscillation.oscillation_reset_angle: 3.14 
    #default is .05
    Oscillation.oscillation_reset_dist: 1.5
    Oscillation.x_only_threshold: 1.5

#### Expected behavior

#### Actual behavior
Oscillation critic is only picking up the 'x_only_threshold'. The 'oscillation_reset_angle' and 'oscillation_reset_dist' params remain at their default values of .2 and .04

Looking at the code, x_only_threshold is handled differently than the rest of the parameters. 

```
  oscillation_reset_dist_ = nav_2d_utils::searchAndGetParam(nh_, "oscillation_reset_dist", 0.05);
  oscillation_reset_angle_ = nav_2d_utils::searchAndGetParam(nh_, "oscillation_reset_angle", 0.2);

  nav2_util::declare_parameter_if_not_declared(nh_,
    name_ + ".x_only_threshold", rclcpp::ParameterValue(0.05));
  nh_->get_parameter(name_ + ".x_only_threshold", x_only_threshold_);

```

It seems like the rest of the params do not expect the critic name to be prepended.  
I had to add a param declaration and modified the searchAndGetParam call to look for 
`name_+ '.<param_name>'
` to get this critic to use params my params specified above.

yathartha3, 2019-09-22:
@Jconn As I understand, you set the params in the nav2_bringup/params/nav2_params.yaml file?
It seems to be a bug, that the name is not prepended. I would ask @bpwilcox to make sure.
If that fixed your problem, can you go ahead and submit a PR? Else, I can make that change.

Jconn, 2019-09-22:
Yes, I was setting it in that params file. I will submit a PR.

bpwilcox, 2019-09-23:
The issue is that those parameters are never declared. There's an unimplemented function in `nav_2d_utils::searchAndGetParam` and it only "gets" a parameter, but doesn't declare, so the values  passed via the yaml file will never take. Good catch, you need to declare the parameter before grabbing it off the node.

bpwilcox, 2019-10-10:
@Jconn Are you still planning to submit a PR on this issue?

Jconn, 2019-10-10:
This is was pushed lower on my queue, sorry @bpwilcox. I will get moving on this in a few days. If you have the bandwidth to make the fix then go for it :)  

I got side-tracked on this when I was working towards a demo and saw that some stuff like PathAlign's `forward_point_distance` might have the same issue

Jconn, 2019-10-13:
@bpwilcox I'm failing to build the nav2 master branch, because I'm failing to build ros2 master for reasons I don't understand and am not interested in looking into right now. So I'm not going to submit a PR. I have to prioritize some other things right now, so I'm going to back out of my commitment. My change is [on my fork](https://github.com/Jconn/navigation2/tree/issues/1154_critic_param_name_fix). 
If you like what you see, you can cherry-pick the commit and make a pr. It's a few param name changes :)

bpwilcox, 2019-10-15:
@Jconn Okay, I'll get to it when I have a moment.

bpwilcox, 2019-11-12:
Closed with #1319

closed 2019-11-12 by bpwilcox