ros-navigation/navigation2 #1155: rviz lagging while navigating

yathartha3, 2019-09-23:
## Bug report
Rviz lagging while navigating. Is anyone else seeing this problem, or have tried with other DDS implementations?

**Required Info:**

- Operating System:
  - OS and version (Ubuntu 18.04)

- DDS implementation:
  - rmw_implementation used Fast-RTPS

#### Actual behavior
[image]

The terminal prints out the "usual" dropped message such as:
[INFO] [rviz]: [signalFailure] Drop message: frame 'base_scan' at time 452.601 for reason(0)
[INFO] [rviz]: [signalFailure] Drop message: frame 'odom' at time 1569260413.444 for reason(0)

SteveMacenski, 2019-09-23:
I usually see this on wifi networks, though from the limited set of data shown its hard to figure out if everything is lagging, or its just TF

yathartha3, 2019-10-01:
Closing now. Will test it and reopen if needed.

closed 2019-10-01 by yathartha3