ros-navigation/navigation2 #1156: GoalDist encourages driving into map obstacles

Jconn, 2019-09-23:
**Required Info:**

- Operating System:
Ubuntu 18.04
- Version or commit hash:
0.2.4

 DDS implementation:
opensplice
#### Steps to reproduce issue
1) enable costcloud topic for easy visualization of critic influences (thank you!)
2) launch map 
3) drive near obstacle 
4) place goal on the other side of the obstacle 
#### Expected behavior
GoalDist does not encourage driving into map obstacles
#### Actual behavior
GoalDist does not consider map obstacles in cost analysis 
#### Additional information
Here is a screenshot of what I'm seeing. I drive really close to the long slender obstacle that is embedded in my map file, and then set a navigation goal on the other side of the obstacle. What I see on the /costcloud topic is that my GoalDist critic encourages me to drive through the obstacle.

This screenshot shows the /costcloud pointcloud overlain onto my map. The costcloud is set to show the intensity of the GoalDist critic. I am assuming that red is low cost and blue is high cost, since my goal is always red for the GoalDist critic. 

[image]

SteveMacenski, 2019-09-23:
Keep in mind that DWB creates an objective score of the best option from a number of sources, so its more than possible that any individual score will encourage something less ideal, but that's why you have many critics to evaluate a path. I dont think of this as a bug, but rather a perfectly find example of a simple heuristic. 

If you feel this is disruptive, you're welcome to set the weird score of the distance critic to 0 so that it will not affect your plans. Otherwise, I think we'd be open to your thoughts on how to better derive a critic to accomplish a similar task to the GoalDist critic that doesn't involve some complex signed distance field

Jconn, 2019-09-23:
Okay, I understand that stance and I'll continue to tune my critic scales to achieve better behavior. I have been wrestling with critic scores for a few days, just today discovered the /cost_cloud topic and its really helping me understand behavior.

closed 2019-09-23 by Jconn