ros-navigation/navigation2 #1160: dwb controller sometimes doesn't preempt

Jconn, 2019-09-25:
## Bug report

**Required Info:**

- Operating System:
Ubuntu 18.04
- Version or commit hash:
version 0.2.4
- DDS implementation:
opensplice

Sometimes when I set a navigation goal through the NavigateToPose action interface, the DWB controller will never pick up changes to the global path made by the ComputePathToPose node. This doesn't reproduce all the time for me.

#### Steps to reproduce issue
- Send NavigateToPose command
- Check to see if the following message is logged:

> "Preempting the goal. Passing the new path to the planner."

#### Expected behavior
DwbController will pick up changes to the global path as shown here:
```
  if (action_server_->is_preempt_requested()) {
    RCLCPP_INFO(get_logger(), "Preempting the goal. Passing the new path to the planner.");
    setPlannerPath(action_server_->accept_pending_goal()->path);
  }

```
#### Actual behavior
No log message about preempting the goal is spat out on my terminal. My costmap critics also do not appear to update costs based on the new path.

#### Additional information
I looked into bt_action_node.hpp, and when I enter this case, the FollowPath action client is not satisfying this conditional:

```
        if (goal_updated_ && (status == action_msgs::msg::GoalStatus::STATUS_EXECUTING)) {
          goal_updated_ = false;
          goto new_goal_received;
        }

```
instead my variables are:
`goal_updated_ = true `and `status = action_msgs::msg::GoalStatus::STATUS_ACCEPTED`

But I am certain that FollowPath is executing something, because my robot is still driving towards the goal pose.

SteveMacenski, 2019-09-25:
Well i think either way if its been accepted or executing your preemption call is valid. I’d approve a PR that changes that condition to either of those two

Jconn, 2019-09-25:
got it, I will do that

MengNan-Li, 2019-09-27:
## Bug report

**Required Info:**

- Operating System:
  - ubuntu 18.04
- Version or commit hash:
  - 0.2.4
- DDS implementation:
  -  Fast-RTPS
- ROS2.0 Version:
  -  dashing patch release 3

#### Steps to reproduce issue

```
1. Starting our robot driver
2. launch navigation2
3. Sending a goal
```

#### Expected behavior
The robot move to the goal and update path when encountering obstacles. 

#### Actual behavior
The "transformed_global_plan" topic path will not be updated as the /path topic is updated.
This task is sometimes successful. In other words, this problem doesn't reproduce all the time for me.

#### Additional information
[image]
[image]

MengNan-Li, 2019-09-27:
As @Jconn , inserting `status = action_msgs::msg::GoalStatus::STATUS_ACCEPTED` can avoid the problem.

closed 2019-09-27 by Jconn