ros-navigation/navigation2 #1162: [ERROR] [rviz2]: InvalidParametersException: Stream size does not match calculated image size in Image

MengNan-Li, 2019-09-25:
## Bug report

**Required Info:**

- Operating System:
  - ubuntu18.04
- Version or commit hash:
  - 0.2.4
- DDS implementation:
  - Fast-RTPS
- ros2 version:
  - source dasing Patch Release 3

#### Steps to reproduce issue

```
ros2 launch nav2_bringup nav2_bringup_launch.py
rviz2 (load map)
```

#### Expected behavior
The rviz2 shows the map.
#### Actual behavior
[INFO] [rviz2]: Stereo is NOT SUPPORTED
[INFO] [rviz2]: OpenGl version: 3 (GLSL 1.3)
[INFO] [rviz2]: Stereo is NOT SUPPORTED
[INFO] [rviz2]: Creating 1 swatches_
[ERROR] [rviz2]: InvalidParametersException: Stream size does not match calculated image size in Image::loadRawData at /tmp/binarydeb/ros-dashing-rviz-ogre-vendor-6.1.1/obj-x86_64-linux-gnu/ogre-master-ca665a6-prefix/src/ogre-master-ca665a6/OgreMain/src/OgreImage.cpp (line 283)
[ERROR] [rviz2]: Failed to create 1 swatches_
[INFO] [rviz2]: Creating 2 swatches_
[ERROR] [rviz2]: InvalidParametersException: Stream size does not match calculated image size in Image::loadRawData at /tmp/binarydeb/ros-dashing-rviz-ogre-vendor-6.1.1/obj-x86_64-linux-gnu/ogre-master-ca665a6-prefix/src/ogre-master-ca665a6/OgreMain/src/OgreImage.cpp (line 283)
[ERROR] [rviz2]: Failed to create 2 swatches_
[INFO] [rviz2]: Creating 4 swatches_
[INFO] [rviz2]: Creating 1 swatches_
[ERROR] [rviz2]: InvalidParametersException: Stream size does not match calculated image size in Image::loadRawData at /tmp/binarydeb/ros-dashing-rviz-ogre-vendor-6.1.1/obj-x86_64-linux-gnu/ogre-master-ca665a6-prefix/src/ogre-master-ca665a6/OgreMain/src/OgreImage.cpp (line 283)
[ERROR] [rviz2]: Failed to create 1 swatches_
[INFO] [rviz2]: Creating 2 swatches_
[ERROR] [rviz2]: InvalidParametersException: Stream size does not match calculated image size in Image::loadRawData at /tmp/binarydeb/ros-dashing-rviz-ogre-vendor-6.1.1/obj-x86_64-linux-gnu/ogre-master-ca665a6-prefix/src/ogre-master-ca665a6/OgreMain/src/OgreImage.cpp (line 283)
[ERROR] [rviz2]: Failed to create 2 swatches_
[INFO] [rviz2]: Creating 4 swatches_
[ERROR] [rviz2]: InvalidParametersException: Stream size does not match calculated image size in Image::loadRawData at /tmp/binarydeb/ros-dashing-rviz-ogre-vendor-6.1.1/obj-x86_64-linux-gnu/ogre-master-ca665a6-prefix/src/ogre-master-ca665a6/OgreMain/src/OgreImage.cpp (line 283)
[ERROR] [rviz2]: Failed to create 4 swatches_
[INFO] [rviz2]: Creating 8 swatches_
[ERROR] [rviz2]: InvalidParametersException: Stream size does not match calculated image size in Image::loadRawData at /tmp/binarydeb/ros-dashing-rviz-ogre-vendor-6.1.1/obj-x86_64-linux-gnu/ogre-master-ca665a6-prefix/src/ogre-master-ca665a6/OgreMain/src/OgreImage.cpp (line 283)
[ERROR] [rviz2]: Failed to create 8 swatches_
[ERROR] [rviz2]: InvalidParametersException: Stream size does not match calculated image size in Image::loadRawData at /tmp/binarydeb/ros-dashing-rviz-ogre-vendor-6.1.1/obj-x86_64-linux-gnu/ogre-master-ca665a6-prefix/src/ogre-master-ca665a6/OgreMain/src/OgreImage.cpp (line 283)
terminate called after throwing an instance of 'Ogre::InvalidParametersException'
  what():  InvalidParametersException: Stream size does not match calculated image size in Image::loadRawData at /tmp/binarydeb/ros-dashing-rviz-ogre-vendor-6.1.1/obj-x86_64-linux-gnu/ogre-master-ca665a6-prefix/src/ogre-master-ca665a6/OgreMain/src/OgreImage.cpp (line 283)
Aborted (core dumped)

#### Additional information

----
## Feature request

#### Feature description

#### Implementation considerations

MengNan-Li, 2019-09-25:
#908

yathartha3, 2019-09-25:
Is it navigation issue or rviz? Can you start rviz independently?

MengNan-Li, 2019-09-25:
@yathartha3 Using the navigation2/nav2_bringup/launch/turtlebot3_world.yaml is ok.

MengNan-Li, 2019-09-25:
I am facing this problem with the map below.
[image]

map.yaml

free_thresh: 0.196
image: map.png
negate: 0
occupied_thresh: 0.65
origin: [-25.132128, -18.565032, 0.0]
resolution: 0.03

yathartha3, 2019-09-25:
@MengNan-Li Thanks. I will look into it.

MengNan-Li, 2019-09-25:
@yathartha3 I found  the log message "/tmp/binarydeb/ros-dashing-rviz-ogre-vendor-6.1.1/obj-x86_64-linux-gnu/ogre-master-ca665a6-prefix/src/ogre-master-ca665a6/OgreMain/src/OgreImage.cpp" pointing ros-dashing-rviz-ogre-vendor-6.1.1. So I just apt remove ros-dashing-rviz-ogre-vendor-6.1.1 and the Rviz2 can load my map now.

yathartha3, 2019-09-25:
Glad you fixed it.
Just FYI, latest ros-dashing-rviz-ogre-vendor is version 6.1.4.

closed 2019-09-25 by SteveMacenski