ros-navigation/navigation2 #1164: [BUG] ActionServer's path in ControllerServer doesn't be affected by use_sim_time

vinnamkim, 2019-09-25:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Version or commit hash:
  - latest master(https://github.com/ros-planning/navigation2/commit/78a0272c585dc575d84ea4ec4d3e58d169c7c622)
- DDS implementation:
  - Fast-RTPS

#### Steps to reproduce issue
Add those lines after https://github.com/ros-planning/navigation2/blob/78a0272c585dc575d84ea4ec4d3e58d169c7c622/nav2_controller/src/nav2_controller.cpp#L165
```
std::cout << action_server_->get_current_goal()->path.header.stamp.sec 
          << "." 
          << action_server_->get_current_goal()->path.header.stamp.nanosec
          << std::endl;
```
Then, launch TB3 simulation with `use_sim_time=true`
```
$(terminal) ros2 launch nav2_bringup nav2_tb3_simulation_launch.py
```

#### Actual behavior
It shows the system time like `157999999.12341234` rather than the simulation time like `80.1234`.

yathartha3, 2019-09-25:
@vinnamkim When I echo the clock from the terminal, I see the sim time correctly. Can you please check that your clock is the sim time?

vinnamkim, 2019-09-25:
> @vinnamkim When I echo the clock from the terminal, I see the sim time correctly. Can you please check that your clock is the sim time?

Well, I mean `action_server_->get_current_goal()->path` has a time stamp with the system time, not the sim time. Do you mean that you checked the time stamp of  `action_server_->get_current_goal()->path`?

orduno, 2019-09-25:
@vinnamkim I've reproduced the bug and can confirm that even when the clock topic shows the correct simulation time, the time stamp on the path is different.

The global planner is the one node which might not have the right clock time -- since it's where the path is created and stamped.

I looked into the launch files and we're correctly setting `use_sim_time` to true by logging the value into the terminal at runtime.

I'm looking into the planner to understand why it's not using the correct time.

orduno, 2019-09-25:
@vinnamkim Could you test #1165 to see if that fixes the bug on your end?

vinnamkim, 2019-09-26:
@orduno LGTM and close this issue.

closed 2019-09-26 by vinnamkim