ros-navigation/navigation2 #1179: Navigation fails after first attempt

orduno, 2019-10-01:
## Bug report

**Required Info:**

- Operating System: Ubuntu 18.04
- Version or commit hash: ec834a5d55f9a7f00f551c147717ea8a9d5e796d
- DDS implementation: FAST-RTPS

#### Steps to reproduce issue
Run simulation
```bash
ros2 launch nav2_bringup nav2_tb3_simulation_launch.py
```
Start the system, provide an initial pose, provide a goal pose, let robot move to goal pose and navigation succeed.

Provide a second goal pose.

#### Expected behavior

Robot moves to the second goal pose.

#### Actual behavior

<details>
<summary> Robot doesn't move towards new goal. However `bt_navigator` returns a succeeded messge and we get some transform error messages.
</summary>
<br>

```bash
[controller_server-7] [INFO] [controller_server]: Reached the goal!
[bt_navigator-10] Do not call canTransform or lookupTransform with a timeout unless you are using another thread for populating data. Without a dedicated thread it will always timeout.  If you have a seperate thread servicing tf messages, call setUsingDedicatedThread(true) on your Buffer instance.Do not call canTransform or lookupTransform with a timeout unless you are using another thread for populating data. Without a dedicated thread it will always timeout.  If you have a seperate thread servicing tf messages, call setUsingDedicatedThread(true) on your Buffer instance.Do not call canTransform or lookupTransform with a timeout unless you are using another thread for populating data. Without a dedicated thread it will always timeout.  If you have a seperate thread servicing tf messages, call setUsingDedicatedThread(true) on your Buffer instance.Do not call canTransform or lookupTransform with a timeout unless you are using another thread for populating data. Without a dedicated thread it will always timeout.  If you have a seperate thread servicing tf messages, call setUsingDedicatedThread(true) on your Buffer instance.Do not call canTransform or lookupTransform with a timeout unless you are using another thread for populating data. Without a dedicated thread it will always timeout.  If you have a seperate thread servicing tf messages, call setUsingDedicatedThread(true) on your Buffer instance.Do not call canTransform or lookupTransform with a timeout unless you are using another thread for populating data. Without a dedicated thread it will always timeout.  If you have a seperate thread servicing tf messages, call setUsingDedicatedThread(true) on your Buffer instance.Do not call canTransform or lookupTransform with a timeout unless you are using another thread for populating data. Without a dedicated thread it will always timeout.  If you have a seperate thread servicing tf messages, call setUsingDedicatedThread(true) on your Buffer instance.Do not call canTransform or lookupTransform with a timeout unless you are using another thread for populating data. Without a dedicated thread it will always timeout.  If you have a seperate thread servicing tf messages, call setUsingDedicatedThread(true) on your Buffer instance.Do not call canTransform or lookupTransform with a timeout unless you are using another thread for populating data. Without a dedicated thread it will always timeout.  If you have a seperate thread servicing tf messages, call setUsingDedicatedThread(true) on your Buffer instance.Do not call canTransform or lookupTransform with a timeout unless you are using another thread for populating data. Without a dedicated thread it will always timeout.  If you have a seperate thread servicing tf messages, call setUsingDedicatedThread(true) on your Buffer instance.Do not call canTransform or lookupTransform with a timeout unless you are using another thread for populating data. Without a dedicated thread it will always timeout.  If you have a seperate thread servicing tf messages, call setUsingDedicatedThread(true) on your Buffer instance.Do not call canTransform or lookupTransform with a timeout unless you are using another thread for populating data. Without a dedicated thread it will always timeout.  If you have a seperate thread servicing tf messages, call setUsingDedicatedThread(true) on your Buffer instance.
[INFO] [bt_navigator]: Navigation succeeded
[bt_navigator-10] [INFO] [bt_navigator]: Begin navigating from current location to (1.53, -0.49)
[bt_navigator-10] Do not call canTransform or lookupTransform with a timeout unless you are using another thread for populating data. Without a dedicated thread it will always timeout.  If you have a seperate thread servicing tf messages, call setUsingDedicatedThread(true) on your Buffer instance.
[INFO] [bt_navigator]: Navigation succeeded

```
</details>

mhpanah, 2019-10-01:
@orduno I just noticed this problem as well.

orduno, 2019-10-01:
@mhpanah There's something wrong with the current bt xml, I've created #1184 to go back to the previous while we figure out the problem.

closed 2019-10-04 by orduno