ros-navigation/navigation2 #1181: RVIZ navigation panel occasionally fails to initialize

orduno, 2019-10-01:
## Bug report

**Required Info:**

- Operating System: Ubuntu 18.04
- Version or commit hash: ec834a5d55f9a7f00f551c147717ea8a9d5e796d
- DDS implementation: FAST-RTPS

#### Steps to reproduce issue
Run simulation
```bash
ros2 launch nav2_bringup nav2_tb3_simulation_launch.py
```
Wait for RVIZ to come up.

#### Expected behavior

The `startup` button on the RVIZ `Navigation2` panel is enabled.

#### Actual behavior

Occasionally, the `startup` button is disabled.

#### Additional Information

I noticed that during normal operation the `startup` button starts as disabled and quickly transitions to enabled. Perhaps a race condition is causing the button not to transition.

I noticed the issue when testing on a single robot but occurs more often when testing on multiple robots.

bpwilcox, 2019-10-01:
The startup button should start as disabled until it receives a response from the lifecycle manager. Since autostart is `false` in that launch script, it should immediately catch a response from the lifecycle manager and transition. @orduno Do you see the service request being sent when you launch in those cases it fails?

orduno, 2019-10-22:
Closing the issue as the race condition was fixed in #1215.

closed 2019-10-22 by orduno