ros-navigation/navigation2 #1185: Controller server process dies after reseting

mhpanah, 2019-10-01:
## Bug report

**Required Info:**

- Operating System:
  -18.04
- Version or commit hash:
  - master
- DDS implementation:
  - Fast-RTPS

#### Steps to reproduce issue

```
ros2 launch nav2_bringup nav2_tb3_simulation_launch.py
from rviz click on Startup
set initial pose from rviz
click on reset
click on startup

```

#### Expected behavior
All the nodes should resume and work
#### Actual behavior
[lifecycle_manager-4] [INFO] [lifecycle_manager]: Configuring controller_server
[controller_server-7] [INFO] [controller_server]: Configuring local controller interface
[controller_server-7] [INFO] [local_costmap.local_costmap]: Configuring
[controller_server-7] [INFO] [local_costmap.local_costmap]: Using plugin "obstacle_layer"
[controller_server-7] [INFO] [local_costmap.local_costmap]: Subscribed to Topics: scan
[controller_server-7] [INFO] [local_costmap.local_costmap]: Initialized plugin "obstacle_layer"
[controller_server-7] [INFO] [local_costmap.local_costmap]: Using plugin "inflation_layer"
[controller_server-7] [INFO] [local_costmap.local_costmap]: Initialized plugin "inflation_layer"
[controller_server-7] [INFO] [controller_server]: Created local controller : dwb_core::DWBLocalPlanner
[controller_server-7] [INFO] [controller_server]: Setting transform_tolerance to 0.200000
[controller_server-7] controller_server: /home/mohammad/OTC_workDir/ros2_ws/src/ros/class_loader/src/class_loader_core.cpp:369: void class_loader::impl::revivePreviouslyCreateMetaobjectsFromGraveyard(const string&, class_loader::ClassLoader*): Assertion `obj->typeidBaseClassName() != "UNSET"' failed.
[rviz2-2] [INFO] [rviz]: [signalFailure] Drop message: frame 'base_scan' at time 56.801 for reason(0)
[ERROR] [controller_server-7]: process has died [pid 26479, exit code -6, cmd '/home/mohammad/OTC_workDir/navigation2/install/nav2_controller/lib/nav2_controller/controller_server --ros-args --params-file /tmp/tmppo3zbq5y                '].

#### Additional information

----
## Feature request

#### Feature description

#### Implementation considerations

mhpanah, 2019-10-01:
@bpwilcox Have you noticed this?

bpwilcox, 2019-10-01:
@mhpanah When is the last time you updated your ros2 ws? This is an old error we saw with the class loader before my upstream PR was merged

mhpanah, 2019-10-01:
@bpwilcox Couple weeks. I will close the issue.

closed 2019-10-01 by mhpanah