ros-navigation/navigation2 #1187: What changes should go into the 0.2.5 release? (End of Sept release)

crdelsey, 2019-10-01:
This is the set of commits that have gone into master since the 0.2.4 release. What commits should go into 0.2.5?

- [ ] a05c9440 2019-10-01 'Carl Delsey' Sphinx Docs - Template Only (#1138)
- [ ] a0a9eb23 2019-10-01 'Steven Macenski' Merge pull request #1176 from SteveMacenski/wait_Recovery
- [ ] 691abe36 2019-10-01 'stevemacenski' using builtin msgs duration
- [ ] ec834a5d 2019-10-01 'Carlos A. Orduno' [multirobot - Part4] Multiple robot control via RVIZ (#1178)
- [ ] 21a3b113 2019-10-01 'stevemacenski' check for negative or zero values in duration
- [ ] e9e96a39 2019-10-01 'stevemacenski' remove unnecessary indentations
- [ ] 2c84c534 2019-09-30 'stevemacenski' adding wait recovery into recovery behavior tree
- [ ] 0d21ddd4 2019-09-30 'stevemacenski' removing extraneous copy of extra bt
- [ ] e17b29ef 2019-09-30 'stevemacenski' wait recovery behavior
- [ ] 4d80d5d3 2019-09-30 'stevemacenski' nav2 recovery interface for bt navigator wait
- [ ] 8e886867 2019-09-30 'Ruffin' Merge pull request #1142 from ros-planning/update_dockerfile
- [ ] 818ae9d8 2019-09-19 'ruffsl' Use ament index to resolve rosdep install
- [ ] 0df1713b 2019-09-30 'Ruffin' Merge pull request #1158 from ros-planning/ruffsl-patch-1
- [ ] ba9cba71 2019-09-30 'bpwilcox' add sleep before checking for collision, remove nav2_dynamic_params from test suite
- [ ] 46b32957 2019-09-30 'Steven Macenski' Merge pull request #1131 from SteveMacenski/nav2_recovery_server
- [ ] 0fa1a95e 2019-09-30 'Carlos A. Orduno' [multirobot - Part3] Add multi-robot launching (#1148)
- [ ] e96d1851 2019-09-30 'ruffsl' Always build underlay as release
- [ ] 180814ba 2019-09-30 'stevemacenski' Merge branch 'master' of github.com:ros-planning/navigation2 into nav2_recovery_server
- [ ] 7063776a 2019-09-17 'Matthew Hansen' Update behavior tree with local and global recoveries
- [ ] fa82367b 2019-09-30 'stevemacenski' fixing backup recovery crash from null node ptr
- [ ] 9517be78 2019-09-27 'Carlos A. Orduno' Fix linter errors (#1174)
- [ ] 50c3157b 2019-09-27 'stevemacenski' adding onCleanup and declaring params in constructor
- [ ] c9d4241a 2019-09-27 'Steven Macenski' Merge pull request #1169 from Jconn/issues/1160_listen_to_goal_changes
- [ ] 90e7a990 2019-09-26 'Steven Macenski' Merge pull request #1171 from shivaang12/doc-recovery-typo
- [ ] d22b681e 2019-09-26 'Shivang Patel' Fixed documentation typo.
- [ ] bcc9fd0d 2019-09-25 'johnconn' prevent dwb controller from missing changes to the global path (#1160)
- [ ] 226f06ce 2019-09-25 'Carlos A. Orduno' [multirobot] Add namespacing to nav stack. (#1147)
- [ ] a0ee02ea 2019-09-25 'Carlos A. Orduno' [multirobot] Spawn multiple robots into Gazebo (#1146)
- [ ] 9c061867 2019-09-25 'Carlos A. Orduno' Fix planner server time bug (#1165)
- [ ] 0bf7f9ce 2019-09-25 'Carlos A. Orduno' Enable deprecated warnings (#1167)
- [ ] c0777fb9 2019-09-25 'bpwilcox' check for always_reset_initial_pose flag in configure step
- [ ] 0828c894 2019-09-23 'bpwilcox' reset initial pose with last known pose covariances
- [ ] 2d368caf 2019-09-25 'Carlos Orduno' Fix typo
- [ ] e40ffaf5 2019-09-25 'stevemacenski' unique ptrs from shared ptrs
- [ ] 78a0272c 2019-09-24 'Melih Erdogan' Navigation 2 with SLAM Documentation (#1143)
- [ ] 1193dbc9 2019-09-24 'Carlos A. Orduno' Update design for providing transform to bt nodes (#1157)
- [ ] 700f3655 2019-09-24 'stevemacenski' shared recovery ptrs to unique
- [ ] e9734e00 2019-09-24 'Ruffin' Add test for rmw_cyclonedds_cpp to nightly job
- [ ] 206105cf 2019-09-24 'stevemacenski' update recoveies_server_node -> recoveries_server & other feedback
- [ ] da9cec63 2019-09-23 'Ruffin' Simplify checksum commands for consistency
- [ ] f4e70486 2019-09-23 'Ruffin' Include complete path of checksum.txt in checksum
- [ ] 555bd7fb 2019-09-23 'Ruffin' Fix underlay parameter for install_overlay_dependencies
- [ ] 8b8985ec 2019-09-23 'Steven Macenski' Update test_system_launch.py
- [ ] 704dbaf3 2019-09-23 'stevemacenski' add memory header to recovery.hpp
- [ ] b7c6761a 2019-09-22 'bpwilcox' fix rebase issues / modify for autostart fix
- [ ] 5f37f6e8 2019-09-10 'bpwilcox' add pause, resume, and repeat button transitions
- [ ] 544e06f3 2019-09-10 'bpwilcox' add pausing/resuming states and slots
- [ ] 5da8e211 2019-09-17 'bpwilcox' fix rebase issue
- [ ] e6092fa6 2019-09-17 'bpwilcox' return system status with timeout
- [ ] 535aba09 2019-09-05 'bpwilcox' lint and uncrustify cleanup
- [ ] 1f28d27a 2019-09-04 'bpwilcox' add service/client for active system and call in Qthread
- [ ] d8ab5d2a 2019-09-20 'stevemacenski' fix merge conflict
- [ ] 0c611bac 2019-09-20 'stevemacenski' merge commit
- [ ] decb36fc 2019-09-20 'Dan Rose' Change "the system" to "managed nodes"
- [ ] 5f96f5b3 2019-09-18 'bpwilcox' create action server on BtNavigator node
- [ ] cc50e2e6 2019-09-19 'bpwilcox' set first_map_received to false when cleaning up
- [ ] 030a0e1d 2019-09-18 'bpwilcox' declare_parameter_if_not_declared in navfnplanner
- [ ] ea0524fb 2019-09-19 'Yathartha Tuladhar' dwb_core.xml
- [ ] 550e1d85 2019-09-17 'Yathartha Tuladhar' Plugin naming cleanup
- [ ] ddb24c55 2019-09-16 'Yathartha Tuladhar' Addressing GCC diagnostic push, pop suggestion
- [ ] c0bc5314 2019-09-14 'Yathartha Tuladhar' 
- [ ] b900d0c4 2019-09-13 'Yathartha Tuladhar' Cleanups and renames
- [ ] 65eb602e 2019-09-13 'Yathartha Tuladhar' Renaming, cleanup, and reintroduced TF buffer
- [ ] 176dc7db 2019-09-12 'Yathartha Tuladhar' Fix linters and delete unused files
- [ ] f5109950 2019-09-12 'Yathartha Tuladhar' Enabling 3D messages for the planner interface
- [ ] 243a8c78 2019-09-12 'Yathartha Tuladhar' Using dwb as plugin
- [ ] 387d0320 2019-09-12 'Yathartha Tuladhar' Creating nav2_controller, and deleting dwb_controller
- [ ] 597c9186 2019-09-12 'Yathartha Tuladhar' nav2_core interface, with lifecycle changes and cleanup
[... 85 lines cut ...]
- [ ] 92a50aa6 2019-09-10 'stevemacenski' updating dwb for removal of nav2_msgs
- [ ] fc9ab5d2 2019-09-10 'stevemacenski' updating global and recovery base type headers
- [ ] 465e4e1d 2019-09-10 'stevemacenski' removing nav2_msgs/path in favor of nav_msgs/Path which inherently contains more information
- [ ] 41da7b80 2019-09-10 'Steven Macenski' Merge pull request #1089 from SteveMacenski/model_navfn
- [ ] 0c5641ff 2019-09-10 'Steven Macenski' Merge pull request #1101 from lopsided98/map-loader-sdl-include
- [ ] 71d5d715 2019-09-10 'stevemacenski' change default state in case of first hit TF failure & hundred test 3 times for similar
- [ ] 4efa7357 2019-09-10 'stevemacenski' removing overloaded collision/tolerance methods
- [ ] 675717da 2019-09-10 'stevemacenski' add activation in node tests
- [ ] 50bb9131 2019-09-10 'stevemacenski' merge conflict resolution from global planner test updates
- [ ] 6993fcd9 2019-09-09 'Carlos A. Orduno' Improvements to global planner tests (#1083)
- [ ] 41eeab2d 2019-09-09 'stevemacenski' updating versions to current 0.2.4
- [ ] 183d6ece 2019-09-09 'stevemacenski' moving cost_value header from navfn_planner.hpp to .cpp
- [ ] b87942a3 2019-09-06 'Ben Wolsieffer' map_loader: add SDL include directories
- [ ] 0514c9aa 2019-09-06 'stevemacenski' removing debug artifacts
- [ ] cba1e8cf 2019-09-05 'bpwilcox' change srv name, remove msg
- [ ] d5f1dcc3 2019-08-29 'bpwilcox' fix lint and uncrustify issues
- [ ] 0363a174 2019-08-29 'bpwilcox' remove reason srv string
- [ ] 5aa6b51b 2019-08-29 'bpwilcox' add pause, resume, reset to manager client
- [ ] f5ef0d06 2019-08-28 'bpwilcox' create custom lifecycle manager srv
- [ ] 94f06fb9 2019-08-28 'bpwilcox' configure and activate all nodes separately
- [ ] 240bcef7 2019-09-05 'Steven Macenski' Merge pull request #1087 from SteveMacenski/nav2_core
- [ ] e18090ba 2019-09-05 'stevemacenski' update licenses in nav2_core to be Apache
- [ ] 41a48739 2019-09-04 'stevemacenski' updating nav_core2 license to apache
- [ ] 78854bdb 2019-09-04 'stevemacenski' plugins receive lifecycle node pointers
- [ ] fb25e8fd 2019-09-04 'stevemacenski' fixed CI issues from incorrect static map state and others

orduno, 2019-10-01:
(I'll be updating this comment as check more items from the list)

Checked items:
- Checking https://github.com/ros-planning/navigation2/commit/1193dbc90690944dd01ccc85736cf3ff021f0f1e as there's an issue reported the bug on Dashing.
- https://github.com/ros-planning/navigation2/commit/0bf7f9ce3eb1862271ccb329d515df653a24f32b is kind of minor, doesn't hut getting in.

Questions:
- We got a long list :fearful:. Can we discuss in general what features we want in? We have:
  - Multi-robot
  - Global planner refactors
  - Local planner refactors
  - Recoveries refactors
  - etc...

crdelsey, 2019-10-02:
> We got a long list. 

Yeah. Tell me about it. I kind of gave up last month on being surgical and just merged everything. That lead to merging something that shouldn't have been merged. I would not have known it wasn't supposed to be merged, though. I think we need the authors, or at least someone familiar with the change to make the call on whether it should be backported. It would be nice if we added a flag to PRs to say whether we want to backport or not, but right now, there's no way to enforce that.

> Can we discuss in general what features we want in?

After discussing a bit yesterday, I'm of the opinion we just want to back port bug fixes to dashing at this point. Eloquent is not that far away, and Dashing is usable, so we don't need to back port features

SteveMacenski, 2019-10-02:
> After discussing a bit yesterday, I'm of the opinion we just want to back port bug fixes to dashing at this point. Eloquent is not that far away, and Dashing is usable, so we don't need to back port features

I think that's typically what OSRF recommends anyhow. I don't necessarily subscribe to that mentality, but there's too much entropy in this project and its not a good use of your time to try to knit-pick features as you've mentioned. :+1: for back porting genuine bugs only

mkhansen-intel, 2019-10-02:
> After discussing a bit yesterday, I'm of the opinion we just want to back port bug fixes to dashing at this point. Eloquent is not that far away, and Dashing is usable, so we don't need to back port features

Seeing this list I agree.

mkhansen-intel, 2019-10-02:
I submitted a PR #1193 directly to the dashing devel branch as a bug fix that needs to be merged before 0.2.5

ruffsl, 2019-10-03:
Any of my commits for patching the CI should be fine, given they don't intersect with the nav2 code. It could help keep the CI consistent when testing PRs for backporting to dashing. Though, I may need to tweek the docker hooks on the dashing branch to switch the executor image from ros2 nightly to dashing.

mkhansen-intel, 2019-10-03:
@ruffsl - can you check the boxes above for the commits you want backported?

> Though, I may need to tweek the docker hooks on the dashing branch to switch the executor image from ros2 nightly to dashing.

It would be great if you could fix the CI on the dashing branch! :+1:

crdelsey, 2019-10-03:
@orduno I'm not going to bring in 0bf7f9c, or else I'd also need to bring in the changes that replace the deprecated functions. I don't think we really need that in the dashing release.

SteveMacenski, 2019-10-10:
@crdelsey #1198 was merged, should I do a release?

SteveMacenski, 2019-10-10:
closing and doing release

closed 2019-10-10 by SteveMacenski