ros-navigation/navigation2 #1194: Getting "Timed out waiting for transform from base_link to map to become available"

shivaang12, 2019-10-02:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Version or commit hash:
  - 
- DDS implementation:
  - default

#### Steps to reproduce issue
nav2 bringup instruction for gazebo.

```

```

#### Expected behavior
Not getting the below error, at least.
#### Actual behavior
[amcl-3] [WARN] [amcl]: Waiting for map....
[planner_server-5] [INFO] [global_costmap.planner_server]: Timed out waiting for transform from base_link to map to become available, tf error: . canTransform returned after 1.00014 timeout was 1.
[planner_server-5] [INFO] [global_costmap.planner_server]: Timed out waiting for transform from base_link to map to become available, tf error: . canTransform returned after 1.00072 timeout was 1.
#### Additional information
I followed the nav2_bringup instruction and couldn't get this thing fixed. Thanks in advance! 

----
## Feature request

#### Feature description

#### Implementation considerations

bpwilcox, 2019-10-02:
Hi @shivaang12, thanks for your issue. Could you provide more information regarding your bringup procedure such as:
- What map are you using? Is it being visualized in rviz or can see it via `ros2 topic echo`?
-  Did you provide a pose in rviz via the "2D Pose Estimate" button?

shivaang12, 2019-10-03:
Hi @bpwilcox,

Sorry for the trouble, I guess I forgot to provide the 2D pose. Thank you for your time.

closed 2019-10-03 by shivaang12