ros-navigation/navigation2 #1200: RVIZ nav2 panel inactive with CycloneDDS

orduno, 2019-10-04:
## Bug report

**Required Info:**

- Operating System:
Ubuntu 18.04
- Version or commit hash:
Nav2 126464caf33a465b9944de5d4a0ea287c6935cfa
- DDS implementation:
Cyclone DDS 378b4b19d7a6e6e3d863f3cb16be8b4d000c2734
Cyclone RMW 1205ab5ea223a72253db9c36b19707978b122a9e

#### Steps to reproduce issue

- First [Build Cyclone](https://github.com/ros2/rmw_cyclonedds). To avoid the build error on Cyclone turn off the IDL preprocessor on the CMake file: `option(BUILD_IDLC "Build IDL preprocessor" OFF)`. This will require to rebuild ros2, nav2, etc.
- Set Cyclone as your DDS implementation to use: `export RMW_IMPLEMENTATION=rmw_cyclonedds_cpp`
- Start the nav stack: `ros2 launch nav2_bringup nav2_tb3_simulation_launch.py`

#### Expected behavior
- RVIZ nav2 panel `start` button is active.

#### Actual behavior
- `start` button doesn't transition to active (remains greyed out).

#### Additional information
- The `start` button is waiting for the `LifecycleManager` to become available, which is, however, it's not receiving the status.

SteveMacenski, 2019-10-17:
Pinging @joespeed, maybe he has thoughts

mkhansen-intel, 2019-10-17:
This is fixed by #1215, not sure why the tool didn't close it automatically when that merged. Closing now.

closed 2019-10-17 by mkhansen-intel