ros-navigation/navigation2 #1201: BT navigator crashes with Cyclone DDS

orduno, 2019-10-04:
## Bug report

**Required Info:**

- Operating System:
Ubuntu 18.04
- Version or commit hash:
Nav2 126464caf33a465b9944de5d4a0ea287c6935cfa
- DDS implementation:
Cyclone DDS 378b4b19d7a6e6e3d863f3cb16be8b4d000c2734
Cyclone RMW 1205ab5ea223a72253db9c36b19707978b122a9e

#### Steps to reproduce issue

- First [Build Cyclone](https://github.com/ros2/rmw_cyclonedds). To avoid the build error on Cyclone turn off the IDL preprocessor on the CMake file: `option(BUILD_IDLC "Build IDL preprocessor" OFF)`. This will require to rebuild ros2, nav2, etc.
- Set Cyclone as your DDS implementation to use: `export RMW_IMPLEMENTATION=rmw_cyclonedds_cpp`
- Start the nav stack: `ros2 launch nav2_bringup nav2_tb3_simulation_launch.py autostart:=true` *You have to set `autostart` to true given #1200.*
- Provide initial pose, wait for stack to become active
- Provide the goal pose.

#### Expected behavior
- Robot start moving towards goal

#### Actual behavior
- `BTNavigator` crashes

```bash
[bt_navigator-10] [INFO] [bt_navigator]: Begin navigating from current location to (1.62, -0.68)
[bt_navigator-10] terminate called after throwing an instance of 'rclcpp::exceptions::RCLError'
[bt_navigator-10]   what():  error taking goal response: error not set
[ERROR] [bt_navigator-10]: process has died [pid 26314, exit code -6, cmd '/home/caorduno/master/overlay_ws/install/nav2_bt_navigator/lib/nav2_bt_navigator/bt_navigator --ros-args -r __node:=bt_navigator --params-file /tmp/tmp_smq99go        
```

orduno, 2019-10-04:
@rotu Are you experiencing #1200 and #1201?

rotu, 2019-10-04:
@orduno I haven't seen these. I'm very curious to get to the bottom of this: `error taking goal response: error not set`

Jconn, 2019-10-04:
This is a duplicate of https://github.com/ros-planning/navigation2/issues/1124, me and @airuchen are hitting that error message

mkhansen-intel, 2019-10-04:
@orduno can you test if the system test passes on your build with cyclonedds?
```
export RMW_IMPLEMENTATION=rmw_cyclonedds_cpp
cd build/nav2_system_tests
ctest -V -R bt_navigator$
```
The system test doesn't use the action interface directly, but uses the 'goal_pose' topic. I've run it 100s of times before with cyclone and it has worked. I'd like to know if it is still working, or if it is also broken. 

Based on a conversation with @crdelsey yesterday, I thought the bt_navigator is creating an internal action client node for the navigateToPose action, in which case I wouldn't expect it to work. So something is strange here.

mkhansen-intel, 2019-10-05:
I tried my own suggestion above and the system test does pass, but when I try to send a goal through the Rviz interface using the `2D Nav Goal` tool, which publshes to the `goal_pose` topic, nothing happens. That's a mystery to me. I don't think I'll have time to look at this until Monday.

Has anyone tried to call this action through python or C++ without using Rviz?

Jconn, 2019-10-05:
I send `2D Nav Goal`s through a standalone python action client, and still see the crash @orduno is reporting

orduno, 2019-10-07:
The crash also occurs if we provide the command directly on the terminal 
``` bash
ros2 action send_goal /NavigateToPose nav2_msgs/NavigateToPose '{pose: {pose: {position: {x: 0.0, y: 0.0, z: 0.0}}}}'
```

orduno, 2019-10-07:
I'm going to close this and follow-up on #1124.

closed 2019-10-07 by orduno