ros-navigation/navigation2 #1206: Improved Recovery Node Behavior Tree

bpwilcox, 2019-10-04:
The current behavior tree and recovery node implement recoveries in a naive way, such that all recovery actions will be executed before the tree retries a `NavigateToPose` action. This also has the consequence that each recovery action must return SUCCESS in order to retry the navigation. This is problematic since if any of the recovery actions fails to complete, it would exit the recovery node and neither execute the remaining recoveries (#1197) nor retry the navigation to the goal pose.

## Feature request
Smarter recovery behavior tree!

#### Feature description
- Allow recoveries to fail and still retry `NavigateToPose` and/or try the next recovery
  - Perhaps there should be a different behavior if the recovery action itself failed?
- Select an appropriate recovery (or set of recoveries) based on a specific error code

#### Implementation considerations
- A "selector" control node that takes in an error code to "select" a recovery would be a great tool for the behavior tree

SteveMacenski, 2019-10-10:
@bpwilcox where does this fall in your priority list? This is something I view as very high priority as an item the navigation1 stack had but we do not and one of the last things to accomplish prior to feature parity.

bpwilcox, 2019-10-10:
@SteveMacenski I agree this is a high priority item, though it is not currently on _my_ plate. @crdelsey is currently porting the behavior tree 3.0 version into the stack and plans to look at these issues for better recovery handling.

SteveMacenski, 2019-10-10:
OK, sorry thought by filing this was something you were thinking about.

SteveMacenski, 2019-11-12:
To summarize the discussion: have more of the contextual recoveries covering the FollowPath/ComputePath that individually try to replan once between them, as well as have the global recovery state try to trigger whatever failed between each recovery.

PR #1388 (2019-12-05): Be able to invoke a single recovery action at a time

---

## Basic Info

| Info | Please fill out this column |
| ------ | ----------- |
| Ticket(s) this addresses   | (fixes: #1206 ) |
| Primary OS tested on | (Ubuntu, 18.04) |
| Robotic platform tested on | (gazebo simulation) |

---

## Description of contribution in a few bullet points

- Addressed #1206 by adding the round-robin node that Mike introduced.
- Round Robin is a custom control type node that can have many children.  If the child returns RUNNING, it will tick the same child until it returns SUCCESS or FAILURE.  If the child re [... cut]
closed 2019-12-05 by mhpanah