ros-navigation/navigation2 #1207: Create a `SimpleActionClient`?

orduno, 2019-10-07:
For discussion:

ROS1 provides a [SimpleActionClient](https://github.com/ros/actionlib/blob/indigo-devel/include/actionlib/client/simple_action_client.h) which is not available on ROS2.

The same situation for `SimpleActionServer`, however, we've created a `nav2_util::SimpleActionServer` (#811).

The SimpleActionClient would wrap `rclcpp:ActionClient` and expose a limited set of features. It would also hide the `GoalHandle` from the user -- user interacts directly with the `SimpleActionClient` to get the result and status of the goal.

mkhansen-intel, 2019-10-07:
Last I knew this was still a "To Do" from this upstream issue:
https://github.com/ros2/ros2/issues/583

I think this discussion should be raised there.

crdelsey, 2019-11-23:
This would be nice to have, but should probably be part of a package outside the nav stack. Closing.

closed 2019-11-23 by crdelsey