ros-navigation/navigation2 #1208: Robot bumps into obstacles not in the static map

orduno, 2019-10-07:
## Bug report

**Required Info:**

- Operating System: Ubuntu 18.04
- Version or commit hash: 07c17a3cd9f8697915a82431b4e8b8cfc8c71506
- DDS implementation: Fast-RTPS

#### Steps to reproduce issue
Launch nav-stack in simulation.
Place an object on the Gazebo environment between the robot and the intended goal pose.
Provide the goal pose.

#### Expected behavior
The robot goes around the obstacle and reaches the goal pose. Alternatively, it could fail if no path is available and trigger a recovery.

#### Actual behavior
The robot goes into the object.

mkhansen-intel, 2019-10-07:
Is anyone looking at this yet? I've reproduced it and started to look into the cause.

bpwilcox, 2019-10-10:
Resolved with #1212

closed 2019-10-10 by bpwilcox