ros-navigation/navigation2 #1216: Lifecycle bringup: homologated list of nodes

SteveMacenski, 2019-10-10:
Right now in the bringup files for navigation vs total bringup there are a list of managed nodes to transition. They are largely the same except localization additions. We should homologate this to a single list to reduce repetition and if you change a boolean for the lifecycle manager you don't change substantive items

crdelsey, 2019-10-14:
@yathartha3 is currently trying to create a nav_bringup_common.py that can contain common code used across the launch files. I'm not sure if it is the best solution for this particular problem, but it is a possible solution.

eg. We could just put the list of node names in the common file `nav2_lifecycle_nodes=['a', 'b', 'c'...]`. Then in each of the launch files, we'd change the lifecycle_manager parameters to something like `'node_names': nav_bringup_common.nav2_lifecycle_nodes`

SteveMacenski, 2019-10-14:
Generally how we did things in ROS1 was to have launch files per package so they were distributed and then 1 bringup package that called each individual package's launch file. That way a 1 liner in the bringup could switch package nodes. 

This is useful especially to separate out launch files for package's libraries that are plugins (ei teb vs dwb).

crdelsey, 2019-10-14:
> Generally how we did things in ROS1 was to have launch files per package

We did discuss doing that back in the early days, but it didn't seem worth it at the time. Most packages would end up with a launch file that just launched that one node. It didn't seem worth the effort over just calling ros2 run for that node. I will admit that our installs are better and ros2 launch is better, making it much easier to include a dependent launch file. It might be worth looking at it again.

That said, the specific problem mentioned in this issue seems like a good example of a problem that can't be pushed down to per-package launch files. Lifecycle manager needs a list of all the nodes being launched in the system. That knowledge only exists at the top level launch file. So, I think a nav2_bringup_common may still be useful here

SteveMacenski, 2019-10-14:
Maybe a dumb question, but why do we care about having a  hardcoded set of current lifecycle nodes to transition? `ros2 lifecycle nodes` will give you all the active lifecycle nodes, I assume there's a C++ version. If we just have the exhaustive life of all nav2 lifecycle nodes, just check who's up now in a lookup, transaction them, and be done. Optionally transition all in case the manager is used for user-level nodes as well. This all assumes the discovery is perfect though. 

> Most packages would end up with a launch file that just launched that one node

I think that's totally fine though. If someone just wants AMCL or Recovery Server, it would be nice to ship with an example launch file (for bringup to orchestrate). Maybe in the future some of these have multiple nodes / components. ROS2 launch also has XML now. 

But none of that is important right now for this ticket. Eventually I'd like the launch files separated so they can be composed more easily for unique cases

crdelsey, 2019-10-15:
> ros2 lifecycle nodes will give you all the active lifecycle nodes

👍  I didn't know you could do that. 

> why do we care about having a hardcoded set of current lifecycle nodes

You solution seems reasonable to me, at least now that @bpwilcox has made it so the nodes can be brought up in any order. An alternative to discovering the nav2 nodes over the network would be to somehow register them within the launch file and have the list of registered nodes passed to lifecycle manager. I'm sure this is doable, but it might not be easy.

SteveMacenski, 2020-01-14:
Taking this on. Mostly its just distributing the task to the appropriate lifecycle manager instances.

SteveMacenski, 2020-01-14:
This is the branch: https://github.com/SteveMacenski/navigation2/tree/simplify-bringup

I'm having a problem reading in the parameters from the parameters file, but I dont think its related to this work, but need to make sure. I've been having that problem on and off since Crystal.

closed 2020-01-31 by SteveMacenski