ros-navigation/navigation2 #1224: Add support for outdoor navigation

crdelsey, 2019-10-11:
#### Feature description
The navigation stack should support navigating in an outdoor environment. This includes features such as:

- [ ] Taking the unevenness of terrain into account in planning.
- [ ] Taking the unevenness of terrain into account when processing sensor data
- [ ] Detecting whether uneven terrain is navigable or not.
- [ ] Handling very large maps
- [ ] Localizing when the nearest features are very far away
- [ ] Fusing AMCL localization with GPS localization

SteveMacenski, 2019-10-11:
The AMCL line is probably removable, if you're really outdoor on uneven terrian, AMCL and 2D mapping in geeneral are poor examples. I think its reasonable to assume positioning is on the userunless we want to get into the 3D slam and localization game in this metapackage

SteveMacenski, 2022-06-03:
https://github.com/ros-planning/navigation2/issues/1278 https://github.com/ros-planning/navigation2/issues/2128 covers this intent for environmental model updates for supporting outdoors better

closed 2022-06-03 by SteveMacenski