ros-navigation/navigation2 #1225: Add support for docking at a charging station

crdelsey, 2019-10-11:
Add support for getting a turtlebot 2 to dock at the Turtlebot 2 charging station. The turtlebot should monitor it's charge level, navigate to the docking station when the charge is low, dock, wait till it's recharged, and then resume its original mission

This serves as an example of how to use the nav2 stack in a larger use case, and also showcases how to swap out the NavFn planner and DWB controller for use-case specific components at run-time.

SteveMacenski, 2019-10-17:
@crdelsey, I like this in concept, but don’t you think this is a little too specific to go in the stack? This supports the turtlebot but not any other robot that doesn’t share the same dock/charging mechanisms. 

For example uses of it, I’d think the ticket for waypoint following would accomplish that. I agree for examples of swapping out controller / planners it would be a good example but I think we could find another sample that could have wider re-use

mkhansen-intel, 2019-10-17:
I think we can make a generic "dock" action interface with a plug-in that allows for the implementation to be robot dependent if needed. We could implement the first one as a Turtlebot2 plugin, and others could use as an example.

crdelsey, 2019-10-21:
>  I like this in concept, but don’t you think this is a little too specific to go in the stack?

Yes. I'm not quite sure what to do with it. Probably it should go in the turtlebot 2 repo or a separate repo. I want the capability, though, so we can do the "robot marathon" test at some point. I want to try to get our turtlebot 2 to drive around the office randomly, and recharge as needed for a day or two or more.

SteveMacenski, 2019-10-21:
If you read the PR2 paper, they plugged in by hand on request. Not saying don’t do it, just an observation

crdelsey, 2019-10-22:
Good to know. I was considering sticking a QR code on the base and then docking using the realsense camera if the built in docking wasn't very good. But, I'll admit, I don't know how hard this will be. Plugging it in by hand would be my second choice.

SteveMacenski, 2019-10-22:
> built in docking

There’s built in docking? If so, why’s this a ticket here?

crdelsey, 2019-10-22:
If the built-in docking works, then all that's required here is to create a behavior tree node that triggers the behavior.

I probably shouldn't have created this issue; it could have just been a note in #929. I'm a bit reluctant to remove it now though since it has several thumbs up from the community. Let's leave it alone for now and see where it ends up on the priority list. We can close it in a month if there isn't much demand for this.

SteveMacenski, 2020-03-02:
This was a particular demo for Intel - closing since I don't have that hardware and the docking controller by all accounts already exists in the turtlebot codebase

closed 2020-03-02 by SteveMacenski