ros-navigation/navigation2 #1226: Support multi-story buildings

crdelsey, 2019-10-11:
Support navigation in multi-story buildings. This involves:

- [ ] Making a planner work across multiple maps
- [ ] Breaking the navigate mission into multiple sub-navigations. ie. nav to elevator, summon elevator, nav into elevator, select next floor, nav to final goal

SteveMacenski, 2019-10-11:
@crdelsey I see this ticket really just asking for2 things:
- ability to change maps in map server
- ability to annotate points with tags in the map

changing maps lets you change floor by a higher level schema. Annotating points lets you say "staging point" and "on elevator" point for each floor map.

As this is a really user specific application, I think those 2 capabilities are what need to be actually introduced and then have sufficient re-usability for other items. With those 2 things implementing this would be trivial + elevator APIs

SteveMacenski, 2019-10-11:
Also #547 and #392 cover this topic

jwallace42, 2022-11-14:
Closing ticket, 

The ability to change maps in in the map server. Annotating the map is current open in #1595.

SteveMacenski, 2022-11-14:
That's not the same as having a solution that's well supported to handle planning and tasks involving multiple floors

jwallace42, 2022-11-14:
Is this a separate solution to the open rmf [traffic editor](https://github.com/open-rmf/rmf_traffic_editor)?

jwallace42, 2022-11-14:
> That's not the same as having a solution that's well supported to handle planning and tasks involving multiple floors

I think that the [open rmf traffic editor](https://github.com/open-rmf/rmf_traffic_editor) supports this ability. Is there something different from this that you had in mind?

SteveMacenski, 2022-11-14:
This is not high on my current list of things to work on or think about (unless there was contributors looking to help out and I would to design if they had the time to implement) so I can't answer in specificity. But until there's a well thought out and fully developed integration with Nav2 and multi-floor navigation, this should stay open.