ros-navigation/navigation2 #1227: Test multiple sensor input support

crdelsey, 2019-10-11:
Support for multiple obstacle sensor sources exists in the obstacle/voxel layer. We need to test and verify it works.

@SteveMacenski There is no support for multiple sensor sources in AMCL. Do you think we need it there as well?

SteveMacenski, 2019-10-11:
AMCL never before supported it, but if you see opportunity to support I wouldn't oppose!

closed 2019-11-05 by SteveMacenski

ninamwa, 2020-02-27:
Is it a requirement that a subscribed topic for the obstacle layer is /scan? I tried testing with two different scan topics, named scan_1 and scan_2. When setting the initial pose in Rviz, nothing happens and the transform from map to odom is never created. When changing the topics to /scan and /scan_2, the initialization is fine again. But then it does look like only the /scan topic is included in the local costmap..