ros-navigation/navigation2 #1228: Documentation improvements

crdelsey, 2019-10-11:
- [ ] Add a top level, getting started page that demonstrates how to run a TB3 example using binaries.
- [ ] Add an architecture overview set of pages
- [ ] Add a parameter reference section
- [ ] Add a section on customizing for a new robot
- [ ] Add a section on parameter tuning.
- [ ] Add a section on building nav2 release branch from source.
- [ ] Add a section on building nav2 master.

SteveMacenski, 2020-02-19:
how those all are structured, their plugin definitions, what messages they send around, relative rates, behavior tree how it works and how we use it, our custom BT nodes, lifecycle manager. 

Behavior tree concept, action server concept, lifecycle manager concept.

SteveMacenski, 2020-03-25:
@gimait if you're so interested, it would be helpful if you could help make some media for the website! The  first two tasks:

```
replace getting started navigation gif so its not terrible.
slam tutorial needs a gif / image
```

Are pretty straight forward with a screen video if you're interested. The goal of these are to have visual examples for all tutorials and pieces of documentation. The SLAM tutorial currently has no gif and the getting started gif at the very end is a poor example of navigation from the very early days (note all the recovery behaviors being triggered and generally struggling). If you could make one that represents the current state of things that aren't poor, that would be awesome and save me a bunch of time (and your videos enshrined into the halls of this project forever).

gimait, 2020-03-25:
For sure, I'll take a look at that :)

gimait, 2020-04-02:
@SteveMacenski could you give me some feedback on the gifs I added in [f3c7c64](https://github.com/gimait/navigation2/commit/f3c7c643ebbf9442937472d1b77283ece52581a3)?
I could change framerate and/or resolution. Is there a size limit I should be aware of for these files?

SteveMacenski, 2020-04-02:
For both, I'd say increase the frame rate a bunch so that its smooth, we don't want it to be choppy. Your recordings are 800x400 while the originals are 600 x 375. Not saying that's necessarily the right resolution, but its big enough to get the point across but small enough to save space. Finally, try also compressing the mp4 before gifing it to decrease size. @mlherd any recommendations on that?

For the navigation: I'd say remove the local costmap since its a little distracting, the global and plan is illustrative enough.

For the slam: Can you increase the laser max distance (temporarily) for this demo? Its a little subtle why only the left 90 degrees is marking. Its not immediately clear to folks that don't do SLAM alot that its because those other beams shown are out of range. Or just have it building the map in a smaller room that it can see more of. The gif only shows a few feet of which very little is actually mapped.

mlherd, 2020-04-02:
I use [ezgif](https://ezgif.com/maker) to create gifs since it has all the tools I need.

-20 fps (15 seconds) should give you good results. Even 10 fps will look good if you really need more seconds. Another option, you can seed up the video a little bit if 15 seconds is not enough.
-Keep the resolution as high as possible ex: 600px

By the way, it seem like the realtime factor drops in Gazebo when tb3 start moving. You can get rid of the camera, depth camera plugins etc in the robot model if you don't use them. It will help you increase the real time factor.

gimait, 2020-04-02:
Awesome! thanks for the help! I’ll push some better gifs soon.

gimait, 2020-04-06:
@SteveMacenski, @mlherd, what do you think about the new gifs?

SteveMacenski, 2020-04-06:
Both great!

mlherd, 2020-04-07:
They both show the expected results, so they are good. The lag between RVIZ and Gazebo is a bit noticeable, but it is fine.

mlherd, 2020-04-07:
> @SteveMacenski, @mlherd, what do you think about the new gifs?

They both show the expected results, so they are good. The lag between RVIZ and Gazebo is a bit noticeable, but it is fine.

gimait, 2020-04-08:
>  The lag between RVIZ and Gazebo is a bit noticeable, but it is fine.

I disabled cameras etc from the robot on gazebo, and although it improved the real time factor, it didn't really help with the lag. Do you have any ideas on how could I fix this?
Otherwise I'll create a pr for the current gifs.

SteveMacenski, 2020-04-08:
I think this is fine - honestly didnt notice it I'm so used to gazebo doing gazebo-y-things

closed 2020-05-04 by SteveMacenski