ros-navigation/navigation2 #1229: F Turtle Release Feature List

crdelsey, 2019-10-11:
This is a list of high level features being considered for the ROS 2 F release. 

Please submit an issue if there is another important feature you'd like to see.
Please give a  :+1: on any issues that you feel are most important. We will try to prioritize issues that have a lot of community interest.

- [ ] Reduce the number of outstanding issues
- [ ] Behavior tree plugin support #958
- [ ] Improve documentation #1228
- [ ] Outdoor navigation support #1224
- [ ] Add support for map zones #404, #401
- [ ] Docking support #1225
- [ ] Support for navigating multi-story buildings #1226
- [ ] Voxel layer support #955
- [ ] Support multiple sensor inputs #1227
- [ ] Support for changing maps #239
- [ ] Add support for multi-layer maps and multiple map sources #392
- [ ] Waypoint follower demo #803
- [ ] Improve user experience (TBD)
- [ ] Ensure a stable, functional SLAM implementation is available for ROS 2 (outside the nav stack)
- [ ] Ensure a stable, functional IMU/Odom fusing solution (eg [Robot Localization] (https://github.com/cra-ros-pkg/robot_localization)) is availble for ROS2. (outside the nav stack)

SteveMacenski, 2019-10-11:
I'd add to the list it is my intention within the next 2 releases to replace navfn entirely for a hybrid approach taking into account robot kinematics and footprint. A special case of that would be a circular holonomic robot which would essentially act as navfn.

orduno, 2019-10-15:
> I'd add to the list it is my intention within the next 2 releases to replace navfn entirely for a hybrid approach taking into account robot kinematics and footprint. A special case of that would be a circular holonomic robot which would essentially act as navfn. This is key for ackermann or non-circular robots

Cool! Let's chat about this, I'm also interested in doing some work on this area.

shivaang12, 2019-10-17:
> I'd add to the list it is my intention within the next 2 releases to replace navfn entirely for a hybrid approach taking into account robot kinematics and footprint. A special case of that would be [...]

Do you mean you want to consider non-holonomic constrains?

SteveMacenski, 2020-03-02:
Moving tickets into the Foxy milestone, so we don't have redundant lists.

See list here: https://github.com/ros-planning/navigation2/milestone/15

closed 2020-03-02 by SteveMacenski