ros-navigation/navigation2 #1240: Make it easier to activate lifecycle nodes started through ros2 run

crdelsey, 2019-10-14:
## Bug report

**Required Info:**

- Operating System:
  - 
- Version or commit hash:
  - 
- DDS implementation:
  - 

#### Steps to reproduce issue
* As discussed in #975, if a user runs a nav2 node directly - eg `ros2 run nav2_map_server map_server`, the node comes up in the unconfigured state. For users unfamiliar with the fact that nav2 uses lifecycle nodes everywhere, this is unexpected behavior, and results in confusion and issues being filed.

#### Additional information

* there are several things we could do to ameliorate the situation:
  1. Make nodes automatically transition to the active state when started via `ros2 run`.
  2. Display a prominent message, indicating that the node needs to be transitioned and how to do it.

We could possibly do this in our code, but, a better solution might be to integrate this into the ros2 run command itself. That way it benefits everyone, not just nav2.

SteveMacenski, 2020-01-13:
To assist, I did make a launch file that automatically transitions into the active state for another project: https://github.com/SteveMacenski/ros2_ouster_drivers/blob/eloquent-devel/ros2_ouster/launch/os1_launch.py

crdelsey, 2020-01-14:
@mjeronimo ran into a bunch of problems trying to get the shutdown path to work when using the LifecycleNode launch action, which is why he wrote lifecycle manager in the first place. However, I think it's time to try to get launch and lifecycle working well together.

In the meantime, for people who didn't use our launch file and just tried to `ros2 run` a node, I've submitted PR #1463

PR #1463 (2020-02-13): Add warning to lifecycle node constructor, so people know they've launched a lifecyle node
## Basic Info

| Info | Please fill out this column |
| ------ | ----------- |
| Ticket(s) this addresses   | fixes #1240 |
| Primary OS tested on | (Ubuntu, MacOS, Windows) |
| Robotic platform tested on | (Steve's Robot, gazebo simulation of Tally, hardware turtlebot) |

---

## Description of contribution in a few bullet points

* Adds a warning to end of each lifecycle node constructor informing the user that they've launched a lifecycle need and need to take addtional action
* This is mainly meant a [... cut]
closed 2020-02-13 by crdelsey