ros-navigation/navigation2 #1244: Provide option/example to launch nav2 as composed nodes

orduno, 2019-10-15:
## Feature request

#### Feature description

#### Implementation considerations

bpwilcox, 2019-10-15:
duplicate of #1013?

orduno, 2019-10-16:
> duplicate of #1013?

I don't think it's a duplicate. If I understand correctly, it's at least a two-step process. I've updated the ticket.

SteveMacenski, 2019-10-16:
I'd consider it one ticket since you really wouldn't know if you did it right without making the launch file to test

regardless, +1000. In the near future I have to finish porting image_pipeline to ROS2 + components so I'll be watching closely to take your lead. 

[image]

SteveMacenski, 2020-08-11:
https://github.com/ros-planning/navigation2/issues/1013 Closing in favor of a single component feature request ticker

closed 2020-08-11 by SteveMacenski