ros-navigation/navigation2 #1245: Multi-robot features

orduno, 2019-10-15:
## Feature request

#### Feature description

This is a meta-issue for new multi-robot feature requests.

Trying to stay within the scope of navigation, here are some ideas:
- [ ] **RVIZ** plugin for multi-robot system visualization and control
- [ ] **Collision avoidance** between robots
- [ ] **Traffic control** and deadlock avoidance
- [ ] **Centralized global path planning**: an entity computes paths for all robots
- [ ] **Aggregated world model**: an entity aggregates sensing data from all robots to build a global world model. Needed for a centralized planner (see above) but there are other possible uses...
- [ ] **Multi-robot coordination**: robots cooperate to execute a mission composed of navigation tasks.

Are you interested in any of these? Do you have other ideas/requests?

shivaang12, 2019-10-16:
Hi @orduno,

I am interested in contributing to these ideas, especially for a centralized global path planning. In the recent past, I have worked on developing global coverage/surveillance planners for multi unmanned aircraft system. May be we can aim something similar here. Let me know your thoughts about it.

orduno, 2019-10-18:
Hi @shivaang12, Thanks for your interest in contributing. I'll touch base with you once we define our priorities for the next release. Hopefully, we can also get more people interested in this topic :smiley:

mattcenoo, 2019-10-21:
Hi @shivaang12, could you list some key methods for solving multi robots navigation problems, and how should it be integrated into the navigation stack. I'm absolutely interested in this topic but still have no idea yet.

shivaang12, 2019-10-24:
Hey @mattcenoo, I just saw this, sorry for replying this late. To answer your question, it depends on what you want from multi robot system. There are many sub areas in this topic and I have experience only with coverage and surveillance part of it. About integration, currently I can see one way we can integrate multi robot system to nav2 by tweaking bt_navigator to provide support for "n robots". But then one more question will arise, you want centralized or decentralized system. So long story short, it is difficult to model multi robot system as generic.

orduno, 2019-10-30:
Users might have a preference for a particular architecture (centralized, distributed, hybrid) depending on the requirements (coherency, fault tolerance, level of coordination, hardware) and constraints for their particular use case. I think we should consider developing an example/solution for each. 

Given that MRS is a huge field, we would have to narrow down the scope and capabilities -- probably to cooperative, strongly coordinated, aware systems.

We can develop a *centralized (or hybrid)* architecture to address industrial applications. Some of the high-level components involved are, at the robot, task execution and, at the orchestrator, task allocation, centralized planning, and mission management. We can do other cool things once we have the *orchestrator*.

On a distributed architecture, we have mostly the task allocation model and coordination protocol. Plenty of solutions have been suggested in the literature, I'm considering a negotiation-based allocation as a starting point.

But, before getting too deep into details, I'd like to start by collecting use cases and an initial set of requirements, where I'd appreciate your input.

We should also think about a set of multi-robot navigation missions, for example, coverage, distribution, etc.

BorgesJVT, 2020-01-27:
Hi @orduno,

I'm also interested in contributing to these features. For the next months I will be working in my thesis, which involves multi-robot coordination and it would be good to do this task integrated with ROS2.

How is the status of these tasks? Is there another channel where these topics are been discussed?

SteveMacenski, 2020-01-27:
@BorgesJVT we have a weekly working group meeting, see calendar here: https://index.ros.org/doc/ros2/Governance/.

The status is that we can simulate multiple robots in multiple namespaces using the TB3 multi-robot launch files. The rest are on-going feature requests and we'd be more than happy to discuss them on the call or find some set of work you could contribute to that effort

SteveMacenski, 2020-03-29:
I found this recently: https://github.com/atb033/multi_agent_path_planning

May be

arpit6232, 2020-05-16:
I would love to pitch in

SteveMacenski, 2020-05-18:
Awesome @arpit6232, any particular area / project to start on?

SteveMacenski, 2021-03-10:
There hasn't been much interest / involvement from folks in awhile around multiple robot features. I'm closing  these tentatively due to this lack of interest. I'm sure if there's groups in the future that would like to consider this, we can open a ticket and scope out the work at that junction. However many of these projects don't very clearly fit in the framework and will have a relatively small impact on the userbase comparatively.

closed 2021-03-10 by SteveMacenski

iLanceS, 2021-11-18:
i need this too