ros-navigation/navigation2 #1248: "Navigation failed" occurs frequently

MengNan-Li, 2019-10-16:
## Bug report

**Required Info:**

- Operating System:
  - ubuntu1804
- Version or commit hash:
  - 0.2.4
- DDS implementation:
  - Fast-RTPS

#### Steps to reproduce issue

```
1. start our robot driver.
2. launch nav2

```

#### Expected behavior
When the robot encounters an obstacle, re-planning will not fail until several attempts.
#### Actual behavior
Sometimes blocking the direction in which the robot is moving, the moving task will fail to execute quickly. This problem can also occur without obstacles, just that there are obstacles to better reproduce this problem.

Log：
[dwb_controller-7] [INFO] [local_costmap.local_costmap]: Received request to clear entirely the local_costmap
[bt_navigator-10] [ERROR] [bt_navigator]: Navigation failed
[dwb_controller-7] [INFO] [dwb_controller]: Goal was canceled. Stopping the robot.

#### Additional information
```
The reason may be that the function below times out.

NavfnPlanner::getCostmap()
auto result = costmap_client_.invoke(request, 5s);
```
#1089 
However, when using the navigation2 master branch, the planner server and controller server processes often crash.

SteveMacenski, 2019-10-16:
I'm trying to dissect your statements a little more.

> re-planning will not fail until several attempts

Do you mean it doesn't fail fast? It will replan through occupied space a few times? Does the costmap reflect those changes? Is this all run on the same computer? Are the update settings reasonable?

> Sometimes blocking the direction in which the robot is moving, the moving task will fail to execute quickly

If there's something blocking it should enter a failure state of executing quickly. I don't see the issue?

```
[dwb_controller-7] [INFO] [local_costmap.local_costmap]: Received request to clear entirely the local_costmap
[bt_navigator-10] [ERROR] [bt_navigator]: Navigation failed
[dwb_controller-7] [INFO] [dwb_controller]: Goal was canceled. Stopping the robot.
```

That looks normal to me. It tries to navigate, couldn't, executed recovery, failed, and gave up.

> auto result = costmap_client_.invoke(request, 5s);
woah, a `5s` timeout, that's not good. Can you provide a link to that line in master / latest dashing?

> often crash
You'd need to know where and when to meaningfully debug

MengNan-Li, 2019-10-16:
> Do you mean it doesn't fail fast? It will replan through occupied space a few times? Does the costmap reflect those changes? Is this all run on the same computer? Are the update settings reasonable?

All Yes.
[image]

> woah, a 5s timeout, that's not good. Can you provide a link to that line in master / latest dashing?

https://github.com/ros-planning/navigation2/blob/2edfb83152f71e771b683c6f834ed1c69b62fbe4/nav2_navfn_planner/src/navfn_planner.cpp#L558

MengNan-Li, 2019-10-16:
https://github.com/ros-planning/navigation2/blob/2edfb83152f71e771b683c6f834ed1c69b62fbe4/nav2_bt_navigator/src/bt_navigator.cpp#L85
https://github.com/ros-planning/navigation2/blob/2edfb83152f71e771b683c6f834ed1c69b62fbe4/nav2_behavior_tree/include/nav2_behavior_tree/bt_service_node.hpp#L64
https://github.com/ros-planning/navigation2/blob/2edfb83152f71e771b683c6f834ed1c69b62fbe4/nav2_behavior_tree/include/nav2_behavior_tree/bt_service_node.hpp#L89
https://github.com/ros-planning/navigation2/blob/2edfb83152f71e771b683c6f834ed1c69b62fbe4/nav2_behavior_tree/src/recovery_node.cpp#L94

Is 10ms enough for clearing global costmap?
Here is the bt xml we use:
[image]

SteveMacenski, 2019-10-16:
Did you change those values and verify that fixes your problem?

MengNan-Li, 2019-10-17:
https://github.com/ros-planning/navigation2/blob/bd0fe8bf238c69bc41ea29680a7a37bd3fb60022/nav2_behavior_tree/include/nav2_behavior_tree/clear_costmap_service.hpp#L39
```
  void on_init() override
  {
        node_loop_timeout_ = std::chrono::milliseconds(100);
  }
```
This can greatly reduce the probability of "navigation failed"

SteveMacenski, 2019-10-17:
Can you give me a plausible explanation why? I'm more than happy to change it to that. I'm just surprised it was a timeout issue there werent additional errors

SteveMacenski, 2019-10-17:
Ah I see, we have a virtual for `on_loop_timeout` that basically no one implements. There should really be a `RCLCPP_WARN` before calling `on_loop_timeout` to mention the fact it timed out.

Can you submit a PR that:
- increases the timeout
- adds a warning stating that the node timed out?

mkhansen-intel, 2019-10-17:
Since this is a blackboard variable, I believe this should be able to be overridden directly in the XML, not sure exact syntax. Something like I think:

```
<ClearEntireCostmap service_name="/global_costmap/clear_global_costmap_entirely" ${node_loop_timeout}="100" >
```
I would bump it to 1000ms though.

mattcenoo, 2019-10-18:
Emm, the blackboard variable can be overwritten using the built-in Actions called _SetBlackboard_,
```<SetBlackboard output_key="node_loop_timeout" value="100" />```
But it'll affect all actiions and services, and since the type of _node_loop_timeout_ is  _std::chrono::milliseconds_, not sure whether this syntax works.

SteveMacenski, 2019-10-18:
I think @mkhansen-intel point was that the parameter can be set on a per-node basis how he showed so you acan choose the specific timeout for the clear costmap service if that's whats only timing out.

Though 10ms seems short even for planning as well

mlherd, 2019-10-18:
I am able to reproduce this issue. I am testing Navigation2 Dashing 0.2.4 on a real robot (Turtlebot2). After sending a goal pose, if there is an obstacle in front of the robot Navigation fails without trying to do any spins or re-planning.

[dwb_controller-7] [INFO] [local_costmap.local_costmap]: Received request to clear entirely the local_costmap
[bt_navigator-10] [ERROR] [bt_navigator]: Navigation failed 

 I will keep investigating the issue.

mattcenoo, 2019-10-18:
Thanks for the quick reply @SteveMacenski. 
Yes, I got what @mkhansen-intel point, but I can't find that syntax after reading though tutorials of BehaviorTree.CPP. And I think the black_board is shared in the entire behavior tree, so I'm not sure how could it be set per-node basis. Correct me if I'm wrong.
I've added a parameter in service node to control the timeout, maybe I could submit a pr if it's appropriate.

SteveMacenski, 2019-10-18:
Id merge that PR!

mlherd, 2019-10-22:
I am able to reproduce this issue using Navigation2 Dashing 0.2.6 on Turtlebot 3. 

I set the time out to 100 ms. 

 ```
void on_init() override
  {
        node_loop_timeout_ = std::chrono::milliseconds(100);
  }
```

I can confirm that this fix reduces navigation failures without executing any recovery behaviors.

SteveMacenski, 2019-10-22:
Its mostly there, we just need a little cleanup to pass CI + place in the behavior tree

closed 2019-10-25 by SteveMacenski