ros-navigation/navigation2 #1252: Figure out why default values in InputPort descriptors are ignored.

crdelsey, 2019-10-17:
The behavior tree library allows one to specify default values to InputPorts, however, that value seems to be ignored when I tested with the RateController object.

If we can get default values working, we can remove the checks on input values in the "NavigateToPose" node.

_Originally posted by @crdelsey in https://github.com/ros-planning/navigation2/pull/1251/review_comment/create_

SteveMacenski, 2019-10-17:
That was also true before in v2. I ran into that fun fact when testing the backup parameterization

shivaang12, 2020-02-27:
@SteveMacenski Can you provide me some more context for this problem? Thanks!

SteveMacenski, 2020-02-27:
Essentially, these defaults (https://github.com/ros-planning/navigation2/blob/master/nav2_behavior_tree/plugins/action/spin_action.cpp#L49) are not being properly used. The "InputPorts" are the inputs that can be taken by a behavior tree node, and there are a number of them. What we have done to get around this is require all XML nodes to fully fill out the inputs or in the setup of the BT node, we try to get it with some bogus number (like duration = -1) and if its still that bogus number on the output, we set to some logical default because clearly it wasnt set (when it was supposed to be) from `getInput` (https://github.com/ros-planning/navigation2/blob/master/nav2_behavior_tree/plugins/action/spin_action.cpp#L41)

shivaang12, 2020-03-02:
Okay on it!

shivaang12, 2020-03-12:
I have opened an issue about this in BehaviorTree.CPP repo here BehaviorTree/BehaviorTree.CPP#153

SteveMacenski, 2020-03-19:
Fixed in BehaviorTree/BehaviorTree.CPP#153

closed 2020-03-19 by SteveMacenski