ros-navigation/navigation2 #1253: Add arbitrary obstacle in CI map that isn't in static map

SteveMacenski, 2019-10-17:
Right now we're not really testing the costmap layers in any meaningful way. We should really have CI run without the static map and ensure the obstacle layer as well as the voxel layer are seeing the obstacles. 

This can be accomplished by changing the gazebo world to include a random block in front of the hardcoded goal. It doesn't have to be moving or anything fancy

mkhansen-intel, 2019-10-17:
Agreed, this is something we've also discussed

mkhansen-intel, 2019-10-17:
Darn you beat me to it @orduno

mlherd, 2019-10-18:
#1186

SteveMacenski, 2019-10-18:
We've not looking for something that's moving (and could collide with the robot on its own accord). We're looking for something to prove the costmap layers are working by having something in the world that isnt in the map. We want to know specifically where the goal is to place this object between the robot and the goal so that if it collides we know there's an issue

mlherd, 2019-10-18:
I was thinking obstacle velocities can be set to 0, but adding an obstacle to the Turtlebot 3 Gazebo world is a simpler solution.

SteveMacenski, 2019-10-18:
Just a big block 2 meters in front of a goal in front of the spawn position would be enough :)

orduno, 2019-10-18:
I'm using the *current* system tests which have a hard-coded start and goal pose (#1243 changes this).

I'm seeing that most of the time (for most obstacle positions between the robot and goal), the robot will trigger recoveries and eventually [fail](https://youtu.be/gNiF5GsJmTM). In just a few locations the robot can actually recover or plan an alternate route and eventually [succeed](https://youtu.be/FcJejE2xqd0). We need to dig deeper but for now, this will suffice.

Note: the goal pose is where I have the cursor on the video and the shortest path goes where the box is placed.

SteveMacenski, 2019-10-18:
@orduno for this test, dont we want static start and stop points? 

What's the benefit of changing to different points? That stops determinism. I'd rather have 1 (or a few) tests that force the robot to do things that prove capability than random generate and you can't say for certain anything in particular works

orduno, 2019-10-18:
Agree, and #1259 uses static end poses.

My reference to #1243 was only because I wanted to adjust the static poses to be closer but didn't want to change was used for other tests.

SteveMacenski, 2019-10-21:
One point for @orduno in the tiny ticket competition!

closed 2019-10-21 by SteveMacenski