ros-navigation/navigation2 #1254: Add Autolocalization as an action server

mkhansen-intel, 2019-10-17:
## Feature request
Auto-localization action server

#### Feature description

As a user, I'd like to be able to tell my robot to "Auto-localize" in some cases. 
1. When it first powers up / starts in the map, instead of having to send an initial pose, I'd like it to try to localize itself
2. When it get's confused or lost (recovery behavior)

I think if the Autolocalization feature was wrapped with an "Autolocalize" action, then the action could be called either from the RViz GUI in case 1 above, or from the Behavior Tree as in case 2 above.
#### Implementation considerations

I believe the autolocalization code is still here in the repo, not sure if it got removed at some point, but it could be forward ported if it was. Adding the action server part shouldn't be difficult (famous last words).

Also, before this wasn't practical because we weren't doing collision checking during backup & spin, now we are so it should be OK to move the robot around a little bit to localize it.

SteveMacenski, 2019-10-17:
Isn't that dependent on not-proven DRL work? I'm not sure that should exist outside the experimental repo yet

mhpanah, 2019-10-17:
> Isn't that dependent on not-proven DRL work? I'm not sure that should exist outside the experimental repo yet

This version is not dependent on the RL work.  In this version, the AutoLocalizer will naively spins and backups in its place to help the particles to converge (it may or may not converge depending on the environment, which in that case the user will know that it failed to converge).  But for the next version, the plan is to have the robot to navigate in such a way to optimally localize itself faster (potentially using RL).

SteveMacenski, 2020-01-18:
@mhpanah @mkhansen-intel What's the status of this given the status of the RL work?

SteveMacenski, 2020-02-04:
I'm closing for now -- if this is premature and there's something to be done, let me know and we can reopen and address what actions we should take.

closed 2020-02-04 by SteveMacenski