ros-navigation/navigation2 #1256: [Tiny Ticket Sprint] 1 ticket award for every 20 methods given doxygen

SteveMacenski, 2019-10-18:
For every twenty (20) methods given doxygen documentation in the headers, I will award 1 ticket worth of points.

Maximum 3 ticket (points). 
If you plan to go this route, I require that you complete entire files and not pick and choose.

SteveMacenski, 2019-10-21:
@shivaang12 has been awarded 1 point from this 

(for my later accounting)

shivaang12, 2019-10-21:
Thanks @SteveMacenski !. I think I will do nav2_controller, if that's fine?

SteveMacenski, 2019-10-21:
Sure thing!

SteveMacenski, 2019-10-22:
@shivaang12 second documentation PR gets them another 1 point!

shivaang12, 2019-10-23:
@SteveMacenski Are we dropping name "Global planner" just like "Local Planner"?

SteveMacenski, 2019-10-23:
I hadn't thought much on global planner, I didn't have an issue with that term. I suppose it could just be "planner" now and that would be fine

shivaang12, 2019-10-23:
Got it! I guess we were using term global just to distinguish it from local planner. Planner sounds cool!

shivaang12, 2019-10-24:
@SteveMacenski I should get one more point?

SteveMacenski, 2019-10-24:
Yup!
You have 3 (maximum) documentation points and are in the lead.

You're also welcome to keep working on documentation or other tickets ;)

shivaang12, 2019-10-24:
Sure!

shivaang12, 2019-10-24:
@SteveMacenski one more question, if you don't mind. Can anyone attend nav2_wg meeting?

SteveMacenski, 2019-10-24:
**Yes**

Resounding yes

closed 2019-10-25 by SteveMacenski

SteveMacenski, 2019-10-26:
@shivaang12 as the winner of the tiny ticket sprint, you may select a graphic that i will place in my ROSCon presentation in China next week!

I'd offer the suggestion to not make my life too hard ;-) 

Please post your selected image to the discourse thread: https://discourse.ros.org/t/ros2-navigations-tiny-ticket-sprint-reward-for-winning/11080 no later than Monday at 6pm so I can get it in (and have a 14 hour flight to figure out what I'm going to do)

shivaang12, 2019-10-27:
@SteveMacenski Thanks man! Could you give me more context into this graphic?

shivaang12, 2019-10-27:
@SteveMacenski And also I don't want to make it difficult for you too. So if you can suggest some topics, it would be great.

SteveMacenski, 2019-10-27:
Just an image, XKCD comic, a robot, I suppose a meme, etc anything you like, that’s appropriate for display. 

If you don’t want to make my life too hard, something graph or laser related since its graph-slam with lasers (who doesnt love lasers?!)