ros-navigation/navigation2 #1271: CircleCI runs colcon test directly instead of run_test_suite script

crdelsey, 2019-10-22:
@ruffsl I noticed that CI is now calling `colcon test` directly instead of using the `tools/run_test_suite`script directly. Any objections to us changing that back?

Calling `colcon test` hides console output so we can't tell why a test is failing in CI. It also doesn't let us run the test a few times to try to get a pass on flaky tests. Given the state of our testing right now, both those things are important.

SteveMacenski, 2019-10-22:
Go for it

ruffsl, 2019-10-23:
Is the test stdout/stderr output from the test not being saved to the colcon log directory? I really hope your not trying to use the nosie stdout of the CI step command UI to debug errors, as apposed to just clicking on the relevant atomic trace log file. That wouldn't be taking advantage of the log artifacts we take effort to save for convenience. 

Also, shouldn't the test scaffolding itself store the test printout to the junit XML files? Many of the tests are missing meta information like timing. We are invoking coloc test directly so that we can parallelize the tests using work scheduling.

If a test is failing, then we should is the colcon filter args like --test-failed in the config to reatemp them, but only the tests that the parallel worker was tasked with.

SteveMacenski, 2019-10-23:
@ruffsl I don't disagree with any of that, but I think more information in the terminal isn't necessarily bad. The parallelism argument is strong, so I'd want to hear about how much this slows it down, but if its not substantial, I don't have a problem with it.

I agree the artifacts are useful for debug but sometimes just having a clear answer in the UI for small stupid errors is nice

ruffsl, 2019-10-23:
Do we have any insight on why the stdout isn't being captured, I'd like to fix the root of the problem rather than a symptom. 

Also, the next Circle I web UI is receiving an update, that folds/box-windows the stdout from steps. This makes it easier to navigate steps over jobs, but will further incentivize the use of artifacts/test-results, as it can hamper reading long nosie prints. 

Quicker visibility into the failing of test should be possible via the large red fail test boxes on the top of the job UI, such that you don't even have to dig into the steps. We just have make sure the test code is reporting about itself correctly.

SteveMacenski, 2019-10-23:
I guess wht I was just saying is: I don't have a problem with it since he asked for objections. I have a workflow that uses the artifacts, but if someone likes their output another way, who am I to judge

ruffsl, 2019-10-23:
What way does this effect stdout and test time? Do we need this in the config:
https://github.com/ros-planning/navigation2/blob/67a92be87f8a00470313c232b1462d45379273e8/tools/ctest_retry.bash#L46

ruffsl, 2019-10-23:
> It also doesn't let us run the test a few times to try to get a pass on flaky tests.

Could we just pass the `--retest-until-pass` arg to colcon, so we don't break the test parallelism:

``` diff
              colcon test \
                --packages-select ${TEST_PACKAGES} \
                --mixin << parameters.mixins >> \
+               --retest-until-pass 3
```

```
$ colcon test --help
...
  --retest-until-pass N
                        Rerun failing tests up to N times
```

https://github.com/ros-planning/navigation2/blob/67a92be87f8a00470313c232b1462d45379273e8/.circleci/config.yml#L142-L156

crdelsey, 2019-10-23:
> Is the test stdout/stderr output from the test not being saved to the colcon log directory?

I didn't know about that. I think that is sufficient for getting the console output.

I can live with the --retest-until-pass 3 for now. We had discussed having different numbers of retries for different tests based on expected failure rate, but I don't think it is very important right now.

@ruffsl Do you know how the test parallelization works? Do the tests run concurrently in the same docker instance, or do they have separate environments? I know some our tests are sensitive to being run concurrently.

crdelsey, 2019-10-23:
> What way does this effect stdout and test time? Do we need this in the config:
 export RCUTILS_CONSOLE_STDOUT_LINE_BUFFERED=1 

This causes ROS2 to output console messages immediately, rather than waiting until a buffer full has been received. Without this, time ordering of console messages is lost.

It will make the test slower, but I doubt it is by much.

Looking at the logs, I believe it is still needed.

ruffsl, 2019-10-23:
> Do you know how the test parallelization works?

Presently, we control test parallelization in the config by changing the `parallelism` key for the job. Although this is currently set to 1, I'd like to retain the ability to simply scale this up in the future as our test suite grows, thus my hestence to break the setup.

https://github.com/ros-planning/navigation2/blob/67a92be87f8a00470313c232b1462d45379273e8/.circleci/config.yml#L375-L382

> Do the tests run concurrently in the same docker instance, or do they have separate environments? I know some our tests are sensitive to being run concurrently.

Parallelism for a CI job is achieved by using separate containers. I also understand that some of our tests are time sensitive or subject to cross talk, thus I elected to use parallelize over containers with separate resources/loopbacks rather than simply resorting to setting `--parallel-workers` in colcon.

The test split itself is done by weighting the split by historic test timing, i.e. a rolling average of time duration for each test, such that we optimize/shorten the total test time via load balancing. That's why it would be nice to fix more of the lengthier tests to correctly report their timing metadata.

https://github.com/ros-planning/navigation2/blob/67a92be87f8a00470313c232b1462d45379273e8/.circleci/config.yml#L144-L150

ruffsl, 2019-10-23:
> Looking at the logs, I believe it is still needed.

@crdelsey , are you

ruffsl, 2019-10-25:
With https://github.com/ros-planning/navigation2/pull/1284 merged, is this closed?

SteveMacenski, 2019-10-25:
I noticed travis is using the script and circleCi is not and this is the reason travis passes with the spawn robot issue, that test isn’t being run in the test suite. 

Do we want to add that, though at the moment that would make Travis fail as well.

crdelsey, 2019-10-30:
> With #1284 merged, is this closed?
Yes. Let's close.