ros-navigation/navigation2 #1274: Active action servers on shutdown aren't going quietly

SteveMacenski, 2019-10-23:
Locally, I'm seeing the CI tests success but on the SIGINT:

[controller_server-7] terminate called after throwing an instance of 'std::runtime_error'
[controller_server-7]   what():  Asked to publish result for goal that does not exist
OR
[bt_navigator-10] [ERROR] [bt_navigator]: Navigation failed
[bt_navigator-10] terminate called after throwing an instance of 'std::runtime_error'
[bt_navigator-10]   what():  Asked to publish result for goal that does not exist

and then causes their respective servers to crash, depending on the situation. Has anyone else noticed this? It seems like when the servers are being brought down, the action servers are still active trying to send an update on terminate that no longer exists. This isn't making the test fail, but that's not a good thing for us to be having going on.

I think there needs to be something protecting the active action servers on exit from trying to `terminate_goal()` or doing anything after the signal has occurred to shutdown so it goes quietly into the night

shivaang12, 2019-10-23:
@SteveMacenski 
I also noticed this same issue on my computer, mine says "navigation failed".

mkhansen-intel, 2019-10-23:
Can you paste in the entire log? Are you sure that the test passed? Keep in mind that sometimes the console output is out of order.

mjeronimo, 2019-11-07:
Is the system being shut down before the processes are terminated? That is, invoke shutdown() service on the lifecycle manager before killing the processes. If this isn't happening the processes are being killed while in the Active state. They should be killed when in the Finalized state, after they've had a chance to stop threads, clean up their resources, etc.

orduno, 2019-11-08:
I think that's the case, I'm working on a PR to solve this.

closed 2019-11-13 by SteveMacenski