ros-navigation/navigation2 #1276: Automatic Parameter Tuning

bpwilcox, 2019-10-23:
## Feature request
Create a tool to automatically tune navigation2 node parameters using state of the art machine learning techniques.

#### Feature description
Currently, the default parameters in the system are some combination of magical numbers, specific to one robot platform (i.e. turtlebot), or may have previous insight. Frankly, there are likely too many parameters in the stack for non-advanced users to tune manually or understand the impact on navigation performance. Dynamic parameter tuning will be achieved soon via dynamic callbacks in the stack (a feature coming soon and already in our milestones), but a tool to automatically tune a subset of parameters would be a potential asset to the stack and various robot platforms/applications. The tool could select a subset of existing navigation2 parameters and through Bayesian optimization techniques create test cases to vary parameters and optimize results. 

#### Implementation considerations
- Use of 3rd party machine learning library?
- ROS2 wrapper for defining test case, optimization criteria, and retrieving results

SteveMacenski, 2019-10-23:
This seems like a large undertaking (even PID tuning is still alot of human in the loop!). I'd first ask for a list of *specific* magic numbers we think this might be good to solve.

I think most things are either geometric or pretty clear what its needed for. The weights of DWB I think is an interesting candidate, implementation details aside.

bpwilcox, 2019-10-23:
> This seems like a large undertaking (even PID tuning is still alot of human in the loop!). I'd first ask for a list of _specific_ magic numbers we think this might be good to solve.
> 
I don't deny this would be a significant task, though I am in the process of scoping this feature now. I would say that almost any parameter that is not a property/state of the robot or environment itself could be a candidate parameter to be tuned. As you mentioned, the controller server parameters are candidates, some in AMCL perhaps too. I think this would be generalize-able to parameter tuning even outside this stack depending on how it is implemented.

SteveMacenski, 2019-10-24:
Sure... usually I’ve seen auto tuning and then a human still tops it off. 

Can you be specific on what parameters in AMCL or costmap layers? Most (if not all?) of these values also have physical property meanings so I’m not sure what weights you’d be autotuning.

Jconn, 2019-10-24:
I think something that could optimize the local costmap resolution and area as a function of robot speed limits, and something that could optimize `vx_samples`, `vtheta_samples` and `sim_time` based on map grid critic resolution, robot critics, and speed limitations would help me, a non-advanced user.

SteveMacenski, 2019-10-24:
The resolution isn't generally related to speed, its related to how close you need to get to objects, how accurate your positioning is, and how much CPU you want to dump into it. 

Similar with the samples -- those are mostly "as many as you can handle in your design spec" on CPU load and how fast you want to run your controller

Jconn, 2019-10-24:
maybe I need to read the documentation, not an auto param tuner :O

When we started missing our controller timing, we tuned the `samples` params and controller rate based on how velocities*sim_time would be binned in the map_grid critics.

SteveMacenski, 2019-10-24:
I can definitely see DWB critic weights and some AMCL weights being tunable, though extremely challenging and data intensive. You can’t really tune amcl in simulation, perhaps DWB you might but defining the spec would be interesting. 

The costmap “stuff” i think is nearly all geometric and you shouldn’t need to tune anything in particular. The local costmap size I suppose could be tunable relative to some CPU load target, but my intuition says that folks’ applications will be the best decider for that (something around moving things will want a larger costmap than others, if youre simulating 10s ahead, you need 10s*max_speed*2 of costmap size at minimum around, etc). 

Again, not saying it doesn’t have its place, but I think a proposal like this needs specific things we can point to to say “tune this” and decide that “yes these are parameters appropriate to be autotuned”.

bpwilcox, 2019-10-24:
I would consider costmap parameters such as `inflation_radius`, `cost_scaling_factor`, `resolution`. For AMCL, I think the `alpha_x` weights, and the `max_particles`, `min_particles` are some candidates as well. We've already mentioned the DWB critic weights, and if we expect the affect on CPU load for some parameters to affect some performance criteria, then they would be candidate parameters as well. 

My thought behind this was to consider the tool as a "mostly" black box optimization and it is up to the user to decide which parameters are of interest to autotune via testing for their context/criteria. I have experience with Bayesian Optimization techniques, so I'm not completely out of my element in proposing this idea.

SteveMacenski, 2019-10-24:
So the alpha_x parameters are motion model and are characteristic of a robot. I expect most professional users do some robot modelling and pick derived numbers. With that said, autotuning could be a way to achieve similar results but those numbers are actually meaningful. 

Min/max particles I could see being tunable as well. In my experience these really are big problems for response time and smoothing tradeoffs. 

Inflation radius is meaningful geometrically to the robot base. Scaling cost could be tunable. Resolution I dont think is necessary to tune.

I think the issue you'll run into in this line of thinking is: how do I generate either ground truth or create metrics to optimize against. For AMCL, you cannot meaningfully tune in simulation since its heavily dependent on real robot characteristics (how off-model is the drive system, wheel slippages, real environment dynamic changes, etc). For DWB, coming up with a metrics to optimize for is an undertaking but conceptually I dont think there's anything blocking you from doing that in simulation. 

Food for thought. In my experience the 2 most challenging and time consuming things to tune are a controller, when you implement a controller the way DWB is, and AMCL. I'd stick with those if going down this line of thinking since the costmap parameters (less cost scale factor) are all geometric. 

My opinion is that this ticket should be closed unless  there's someone specific that's interested and actually going to commit to implementing this. Otherwise its kind of a wishlist item that's going to sit around

bpwilcox, 2019-10-24:
I don't think an argument per parameter is really the point of this ticket, as we've already illuminated several parameters which could be "tunable". I am also the person who has an interest in implementing (or at minimum scoping out the needed work for a formal design proposal) this feature, especially as I see value in said tool not simply in the navigation2 stack but for ros2 in general. I wouldn't close this ticket as it should be part of the discussion on future features for the stack. I posted this issue at request of @crdelsey for this exact purpose, but I appreciate your thoughts.

SteveMacenski, 2020-06-29:
Closing from lack of movement / interest from community. This would be a nice add, but I don't think its realistic to expect this to be ever worked on

closed 2020-06-29 by SteveMacenski

marcusvinicius178, 2024-03-05:
Hello @bpwilcox , I'm highly interested in using and collaborating on a tool that optimizes and fine-tunes both local and global controls based on user needs. Specifically, for the Nav2 stack utilized with GPS to follow waypoints, the process of adjusting various parameters like those for the [MPPI Controller](https://navigation.ros.org/configuration/packages/configuring-mppic.html) MPPI controller and the [Smac Global Planner (https://navigation.ros.org/configuration/packages/smac/configuring-smac-hybrid.html), followed by conducting numerous simulations to achieve successful waypoint navigation, can be quite tedious. A tool that offers pass/fail tests to determine if the robot met its goal, alongside performance metrics such as total **time taken and trajectory distance,** would significantly enhance efficiency. Such a tool, akin to those from MoveIt and Apollo, which allows for parameter **trade-offs and tuning while providing comprehensive feedback on performance** (Pass or Fail), would be invaluable. I'm eager to discuss potential collaboration on this implementation. Please contact me at marcusvini178@usp.br.

bpwilcox, 2024-03-06:
Hi @marcusvinicius178 , I'm glad you are interested in this issue. I unfortunately have not been able to commit resources to develop this kind of tool, though I did draft some initial ideas that we can chat about. I wanted to comment here to state that this type of tool may be better served to live in its own repository outside of the navigation2 repo.