ros-navigation/navigation2 #1281: Travis CI times out when building gazebo_plugins

mkhansen-intel, 2019-10-23:
# Bug report

Travis CI is timing out at 10 minutes due to no output on the console when building gazebo_plugins.

See issue filed here for details:
https://travis-ci.community/t/help-please-how-to-deal-with-a-travis-timeout-in-the-middle-of-a-multi-package-build/5647

crdelsey, 2019-10-23:
We could try passing through the console output when we build that package. Presumably, individual files build in less time than 10 minutes.

eg. After line 65, we can add `--event-handlers console_direct+ \`
https://github.com/ros-planning/navigation2/blob/67a92be87f8a00470313c232b1462d45379273e8/Dockerfile#L61-L65

mkhansen-intel, 2019-10-25:
This is now fixed by #1287

closed 2019-10-25 by mkhansen-intel