ros-navigation/navigation2 #1283: CI Broken: BT XML paths & spawn_robot

SteveMacenski, 2019-10-23:
All CI is failing on a change to the test bringups getting XML path & not being able to find a spawn_robot package:

```
15: [spawn_robot-2] Traceback (most recent call last):
15: [spawn_robot-2]   File "/opt/overlay_ws/install/spawn_robot/lib/spawn_robot/spawn_robot", line 11, in <module>
15: [spawn_robot-2]     load_entry_point('spawn-robot', 'console_scripts', 'spawn_robot')()
15: [spawn_robot-2]   File "/usr/local/lib/python3.6/dist-packages/pkg_resources/__init__.py", line 489, in load_entry_point
15: [spawn_robot-2]     return get_distribution(dist).load_entry_point(group, name)
15: [spawn_robot-2]   File "/usr/local/lib/python3.6/dist-packages/pkg_resources/__init__.py", line 2852, in load_entry_point
15: [spawn_robot-2]     return ep.load()
15: [spawn_robot-2]   File "/usr/local/lib/python3.6/dist-packages/pkg_resources/__init__.py", line 2443, in load
15: [spawn_robot-2]     return self.resolve()
15: [spawn_robot-2]   File "/usr/local/lib/python3.6/dist-packages/pkg_resources/__init__.py", line 2449, in resolve
15: [spawn_robot-2]     module = __import__(self.module_name, fromlist=['__name__'], level=0)
15: [spawn_robot-2] ModuleNotFoundError: No module named 'spawn_robot'
15: [ERROR] [spawn_robot-2]: process has died [pid 5207, exit code 1, cmd '/opt/overlay_ws/install/spawn_robot/lib/spawn_robot/spawn_robot --robot_name robot1 --robot_namespace robot1 --sdf /opt/overlay_ws/src/navigation2/nav2_system_tests/models/turtlebot3_waffle/model.sdf -x 0.0 -y 0.5 -z 0.01 --ros-args'].
15: [controller_server-9] [INFO] [robot1.controller_server]: Created local controller : dwb_core::DWBLocalPlanner
15: [spawn_robot-3] Traceback (most recent call last):
15: [spawn_robot-3]   File "/opt/overlay_ws/install/spawn_robot/lib/spawn_robot/spawn_robot", line 11, in <module>
15: [spawn_robot-3]     load_entry_point('spawn-robot', 'console_scripts', 'spawn_robot')()
15: [spawn_robot-3]   File "/usr/local/lib/python3.6/dist-packages/pkg_resources/__init__.py", line 489, in load_entry_point
15: [spawn_robot-3]     return get_distribution(dist).load_entry_point(group, name)
15: [spawn_robot-3]   File "/usr/local/lib/python3.6/dist-packages/pkg_resources/__init__.py", line 2852, in load_entry_point
15: [spawn_robot-3]     return ep.load()
15: [spawn_robot-3]   File "/usr/local/lib/python3.6/dist-packages/pkg_resources/__init__.py", line 2443, in load
15: [spawn_robot-3]     return self.resolve()
15: [spawn_robot-3]   File "/usr/local/lib/python3.6/dist-packages/pkg_resources/__init__.py", line 2449, in resolve
15: [spawn_robot-3]     module = __import__(self.module_name, fromlist=['__name__'], level=0)
15: [spawn_robot-3] ModuleNotFoundError: No module named 'spawn_robot'
```

This commit https://github.com/ros-planning/navigation2/pull/1269/commits/a8fece364864c3a32758d32fba758eabfe37bcf4 fixes the XML BT paths, but the spawn_robot error is still around

crdelsey, 2019-10-23:
The BT XML stuff looks like a merge error. I believe your commit above is the correct fix. I'm not sure about the spawn_robot issue.

SteveMacenski, 2019-10-23:
@crdelsey can you review the multi planner PR so we can get that part fixed? 

The spawn robot I dont know too much about to debug too far

orduno, 2019-10-24:
I push #1293 to fix the bt xml paths.

I'm not able to reproduce the `spawn_robot` pkg finding issue...

SteveMacenski, 2019-10-24:
@orduno wait for CI to run, you'll see it

SteveMacenski, 2019-10-24:
@orduno reopening, you #1293 PR also exhibits the failing to spawn

```
    [spawn_robot-3] Traceback (most recent call last):
    [spawn_robot-3]   File "/opt/overlay_ws/install/spawn_robot/lib/spawn_robot/spawn_robot", line 11, in <module>
    [spawn_robot-3]     load_entry_point('spawn-robot', 'console_scripts', 'spawn_robot')()
    [spawn_robot-3]   File "/usr/local/lib/python3.6/dist-packages/pkg_resources/__init__.py", line 489, in load_entry_point
    [spawn_robot-3]     return get_distribution(dist).load_entry_point(group, name)
    [spawn_robot-3]   File "/usr/local/lib/python3.6/dist-packages/pkg_resources/__init__.py", line 2852, in load_entry_point
    [spawn_robot-3]     return ep.load()
    [spawn_robot-3]   File "/usr/local/lib/python3.6/dist-packages/pkg_resources/__init__.py", line 2443, in load
    [spawn_robot-3]     return self.resolve()
    [spawn_robot-3]   File "/usr/local/lib/python3.6/dist-packages/pkg_resources/__init__.py", line 2449, in resolve
    [spawn_robot-3]     module = __import__(self.module_name, fromlist=['__name__'], level=0)
    [spawn_robot-3] ModuleNotFoundError: No module named 'spawn_robot'
    [controller_server-9] [INFO] [robot1.local_costmap.local_costmap]: Initialized plugin "inflation_layer"
    [spawn_robot-2] Traceback (most recent call last):
    [spawn_robot-2]   File "/opt/overlay_ws/install/spawn_robot/lib/spawn_robot/spawn_robot", line 11, in <module>
    [spawn_robot-2]     load_entry_point('spawn-robot', 'console_scripts', 'spawn_robot')()
    [spawn_robot-2]   File "/usr/local/lib/python3.6/dist-packages/pkg_resources/__init__.py", line 489, in load_entry_point
    [spawn_robot-2]     return get_distribution(dist).load_entry_point(group, name)
    [spawn_robot-2]   File "/usr/local/lib/python3.6/dist-packages/pkg_resources/__init__.py", line 2852, in load_entry_point
    [spawn_robot-2]     return ep.load()
    [spawn_robot-2]   File "/usr/local/lib/python3.6/dist-packages/pkg_resources/__init__.py", line 2443, in load
    [spawn_robot-2]     return self.resolve()
    [spawn_robot-2]   File "/usr/local/lib/python3.6/dist-packages/pkg_resources/__init__.py", line 2449, in resolve
    [spawn_robot-2]     module = __import__(self.module_name, fromlist=['__name__'], level=0)
    [spawn_robot-2] ModuleNotFoundError: No module named 'spawn_robot'
    [ERROR] [spawn_robot-3]: process has died [pid 5219, exit code 1, cmd '/opt/overlay_ws/install/spawn_robot/lib/spawn_robot/spawn_robot --robot_name robot2 --robot_namespace robot2 --sdf /opt/overlay_ws/src/navigation2/nav2_system_tests/models/turtlebot3_waffle/model.sdf -x 0.0 -y -0.5 -z 0.01 --ros-args'].
    [ERROR] [spawn_robot-2]: process has died [pid 5218, exit code 1, cmd '/opt/overlay_ws/install/spawn_robot/lib/spawn_robot/spawn_robot --robot_name robot1 --robot_namespace robot1 --sdf /opt/overlay_ws/src/navigation2/nav2_system_tests/models/turtlebot3_waffle/model.sdf -x 0.0 -y 0.5 -z 0.01 --ros-args'].
```

closed 2019-10-26 by crdelsey

SteveMacenski, 2019-10-26:
@crdelsey can you give a quick summary of what that was? Given it didn’t appear locally that was a unique thing I haven’t run into before.

PR #1293 (2019-10-24): Fix BT xml paths on tests
---

## Basic Info

| Info | Please fill out this column |
| ------ | ----------- |
| Ticket(s) this addresses   | #1283 |
| Primary OS tested on | Ubuntu 18.04 |
| Robotic platform tested on | Gazebo simulation of TB3 |

---

## Description of contribution in a few bullet points

Path to the BT xml description file was not provided correctly to the tests. Fixes #1283.
PR #1306 (2019-10-26): Get spawn_robot installed correctly - Fix for multi-robot test
## Basic Info

| Info | Please fill out this column |
| ------ | ----------- |
| Ticket(s) this addresses   | fixes #1283 |
| Primary OS tested on | Ubuntu 18.04 |
| Robotic platform tested on | NA |

---

## Description of contribution in a few bullet points

* Minor tweaks to package structure to match other ros python packages.