ros-navigation/navigation2 #1289: BT Action Node Result / Goal logic

SteveMacenski, 2019-10-24:
This is probably not good to be doing, we're even using what I consider to be the 'right toolset' already. We should be while looping over get_status() and when the status is STATUS_SUCCEEDED then we spin the future to get the result.

Right now we're continuously timing out a future request for a result from an action server we **know** isn't done yet, because we're even checking `get_status()` to know if there's a new goal available.

The new workflow should be:

```
while (...) 
{
  get_status()
  if status means do something {
    do that thing
  } else if status means done {
    spin future to get result
    break
  }
}
```

SteveMacenski, 2019-10-24:
Working on it

crdelsey, 2019-11-18:
@SteveMacenski Does #1298 fix this? Can we close?

SteveMacenski, 2019-11-18:
No. #1298 wasn't merged. 

The action clients need to be registering callbacks or using lambdas to get their result and feedback and not polling a future spin.

SteveMacenski, 2019-12-04:
https://github.com/ros-planning/navigation2/pull/1406

crdelsey, 2019-12-04:
So near as I can tell, these are the commits that went in between before the merge of #1384.  Presumably, one of these needs to be the conflict that change.
```
b4c34d365d33c5cb687d49b8336e2f8a8728841b using ClientT for ActionClient typedef
44f9d3a77be765e345cd22a6613834210aa524dd Invert logic for out of bounds check
85a2ac17321b53ddd276eb3f9f4bdbf7be69de13 Correct CamelCase with snake_case action server names (#1385)
0154951f9cbeab9b5272f36ed02d65b25fdda0f4 changing references from local planner to controller outside of DWB (#1377)
d8f13e2d974c24fdce6d6a7f2e349b8c62492b47 Merge pull request #1381 from harderthan/patch-1
992f706936694cd6998e8f476d4432b2e87c6128 Merge pull request #1331 from SteveMacenski/b_to_w
cf5a548ab9dd8c4715c9998a3d74aff6a8f5ca7e Merge pull request #1382 from SteveMacenski/1239
f07680907901504c1a85e901a067a00c47b2adf8 Merge pull request #1383 from SteveMacenski/903
4ece520ba87e1a813fb6d5dc680c01dbb94873af Fix parallel replanning in the behavior tree (alternate implementation) (#1371)
b229f3540de8c764b9481264ff6bd927668c6050 Merge pull request #1361 from SteveMacenski/reorg_bt_nav
753aac95ef733142cd712e2037e461eccc7f8bfa Merge branch 'master' of github.com:ros-planning/navigation2 into reorg_bt_nav
0cb56581f998225a99ec5eabc7c51eb7d0603680 adding warning when robot is out of costmap bounds
451f9aba112aab116595f43603397476c43124a8 move collision checker as an input to the configure method of recoveries
80f6d5ced1fd069b1011c5058fe4c1f47ae80275 Update index.rst
7ed5c3eddf2265b8f6f396f72a5840bac77b741c Update index.rst
```

I suppose it's also possible an upstream change conflicts with this. I guess to prove or disprove this, we could recreate the PR with the original branch. It should pass. Then, if we rebase the branch it should fail if the conflict is in our codebase.

SteveMacenski, 2019-12-05:
https://github.com/ros-planning/navigation2/pull/1415

closed 2020-03-18 by SteveMacenski