ros-navigation/navigation2 #1297: Sperating Astar and Dijkstra from nav2_navfn_planner

shivaang12, 2019-10-24:
----
## Feature request
Dissolving nav2_navfn_planner into Astar and Dijkstra.
#### Feature description
- We'll have two different planner, Astar and Dijkstra, if we want to add more planners in our inventory.

#### Implementation considerations
- Create two different planner one for nav2_Astarfn_planner and another nav2_Dijkstrafn_planner.
- As we'll have two planner we can be put either one of them as default, just as nav2_navfn_planner is default if no other planner specified.

- I think this approach would make more organized and would be more easier later on if we want to add more planner in our inventory.

SteveMacenski, 2019-10-24:
I'm confused, why do you want this?

Navfn lets you select from a parameter which to use

shivaang12, 2019-10-24:
I agree we have both the planner in out inventory and nav2_navfn_planner is great. Just that it would me more organized than "two planner in one package". If you think otherwise, I am good.

SteveMacenski, 2019-10-24:
They just share so much code that separating them out would be a headache to do cleanly.

With that in mind I don't have a problem with it, I'm trying to understand your needs to require it. If this interests you, I'd merge a given PR for it

shivaang12, 2019-10-24:
No worries, I think this might be not that important for now. I look on different task to solve. Thank you for your time. Closing this ticket!

SteveMacenski, 2019-10-24:
Well lets leave it open, maybe someone else agrees and wants to work on it.

mkhansen-intel, 2019-10-24:
I actually agree with you @SteveMacenski. We have visited this topic before and decided the NavFn name with two parameters was the most efficient and somewhat consistent with ROS Navigation so users would understand it.

@shivaang12 - thanks for offering to help, we would definitely welcome you to fix another issue in the backlog.

SteveMacenski, 2019-10-24:
Sounds like we have a decision

closed 2019-10-24 by SteveMacenski