ros-navigation/navigation2 #1300: Alternate interface for Multiple Planner Plugins

crdelsey, 2019-10-25:
As designed in #1269, there is one Action interface per planner/controller server instance.

Let's say we want 3 controllers enabled in the system simultaneously, TEB, DWB and a reinforcement learning based controller. Let's also assume TEB and DWB can share a costmap, but RL works fundamentally differently so needs a different costmap setup.

We can use ROS remapping rules to change the action interface names on each server so we can access them both.

That situation looks something like this:
[image]

The XML in the behavior tree to access each of the different types of plugins will look like this:
```XML
<FollowPath path="{path}" controller="TEB" topic="/Controller2/FollowPath"/>
<FollowPath path="{path}" controller="DWB" topic="/Controller2/FollowPath"/>
<FollowPath path="{path}" controller="" topic="/Controller1/FollowPath"/>
```
 
In this case we need two parameters in the `FollowPath` behavior tree node to direct it to the correct controller plugin. 

We can't use ROS remapping on bt_navigator, because all the FollowPath nodes use the same default `/FollowPath`. There's no way to specify on the command line that the first two FollowPath nodes get remapping `/FollowPath:=/Controller2/FollowPath` and the third gets `/FollowPath:=/Controller1/FollowPath`. It has to be done through XML parameters. 

Also, in this case, deployment details have leaked into the behavior tree XML. If, at some point, you decide that you really want to add the RL controller into the same controller server, you need to change your XML. 

-------

Instead, if each plugin had it's own action interface topic like this:
[image]

The XML looks like this:
```XML
<FollowPath path="{path}" topic="/TEB/FollowPath"/>
<FollowPath path="{path}" topic="/DWB/FollowPath"/>
<FollowPath path="{path}" topic="/RL/FollowPath"/>
```

In this case, there is only one extra parameter to add in the BT XML. And, we can move plugins between controller servers with no change to the XML. In is invariant despite deployment changes.

This is much more complicated to implement, but I believe it to be a clean interface.

crdelsey, 2019-10-25:
Here's a description of what I was trying to convey in the WG meeting yesterday.

SteveMacenski, 2019-10-25:
I think defining the `server` as another field in the behavior tree sounds reasonable (you have `topic`, I'd probably choose `server` or `server_name`), where the default value is the default server we provide `/ComputePathToPose`. So this would only be a tag required if you were doing a non-default multi-planning-server implementation. The convention we could define would be that the `server_name="my_remote_server"` would be tacked onto `ComputePathToPose` to create `my_remote_server/ComputePathToPose`. If the remote server is a `nav2_planning_server`, it could be remapped or take in a namespace parameter on instantiation. In the case of a non-`nav2_planning_server` (ei someone's python action server hosting their RL planner) then they'd have to follow the convention or create their own BT node plugin. Or we could allow them to specify the entire action server path as you suggest `my_remote_server/ComputePathToPose` where the default is `/ComputePathToPose`. I think either are reasonable. 

I believe this would be simple to implement. 

- Additional servers either launched with a remap, or I add a parameter (recommend remap, this is the use-case it was made for, multiple instances of 1 node).
- New servers either follow a convention or we allow for the BT to take in full paths (recommend full path, less documentation required).
- With no server is defined, it defaults to what most users are going to be using (if not using this feature, you never know the difference).

Any issues with that? I can get started after these are in.

crdelsey, 2019-10-25:
>This: controller="TEB" is not what we agreed to and not the intention. controller_name: FollowPath that internal to the server contains a TEB planner is what we agreed to.

Ok. I'm assuming the mapping of the name to the plugin will be configurable? So someone could load the TEB plugin and call it "FollowPathSlowlyAndCarefully" or they could just call it "TEB" if they want to be literal?

> I think defining the server as another field in the behavior tree sounds reasonable (you have topic, I'd probably choose server or server_name), where the default value is the default server we [...]

That's fine with me

> New servers either follow a convention or we allow for the BT to take in full paths (recommend full path, less documentation required).

Just to be clear, for the recommended option, this means having the user create a custom BT node? Or are they still using the FollowPath/ComputePathToPose node but can specify either:
1.  just a server name, or
2. a full path

SteveMacenski, 2019-10-25:
> Ok. I'm assuming the mapping of the name to the plugin will be configurable? So someone could load the TEB plugin and call it "FollowPathSlowlyAndCarefully" or they could just call it "TEB" if they want to be literal?

That is correct. 

> Just to be clear, for the recommended option, this means having the user create a custom BT node? Or are they still using the FollowPath/ComputePathToPose node but can specify either:

The latter, I will implement it so they specify the server name or the full path, its up to us to decide which we support. I like full path so there's no hidden design knowledge. No custom BT node required

closed 2019-12-05 by SteveMacenski