ros-navigation/navigation2 #1301: NavToPose Preemption Broken

orduno, 2019-10-25:
## Bug report

**Required Info:**

- Operating System: Ubuntu 18
- Version or commit hash: f409ba184e54f919bb8a2cbb6d0057b8b3658dde
- DDS implementation: Fast-RTPS and CycloneDDS

#### Steps to reproduce issue
- Start nav stack and provide a goal pose.
- Before goal pose is reached, provide a new goal pose.

#### Expected behavior
- Nav stack computes a path to the new goal pose and robot start following new path.

#### Actual behavior
- Robot continues on the previous path. No new path is displayed on RVIZ and no log info about controller preempting goal:

```bash
[bt_navigator-10] [INFO] [bt_navigator]: Received goal preemption request
[bt_navigator-10] [INFO] [bt_navigator]: Begin navigating from current location to (0.51, 0.50)
```

#### Additional information

yathartha3, 2019-10-28:
Related to #1307 where path not being recalculated at 1 Hz, at thus not re-planning around obstacles once it starts moving on a path.

mattcenoo, 2019-10-30:
Hi, here is something confusing me and may affect the behavior of BT.
(a) If we use the dashing version bt xml, the subnodes of the sequence node will be executed parallelly, and the path will be recalculated at 1 Hz.
[image]
(b) But if we use the master version, the path will not be recalculated at 1 Hz as @yathartha3 mentioned at #1307.
[image]
The question is, what is the exact behavior of a sequence node? I mean, the result of (a) is what we want but is not the expected behavior of a sequence node.

yathartha3, 2019-10-30:
@mattcenoo The exact behavior of what a sequence node should do is described [here](https://behaviortree.github.io/BehaviorTree.CPP/SequenceNode/).

I do not think the sequence node should be running it's subnodes in parallel. How did you verify this? I am really curious to know what is actually happening.

Also, if you are launching `nav2_bringup_launch.py` in master, the default behavior tree xml is `navigate_w_replanning_and_recovery.xml`

mattcenoo, 2019-10-30:
Yes, so do I. To be specific, I'm testing the newest release of dashing-devel (merged some bug fix), here is the comment in the xml.
[image]
BT replans the global path at 1hz indeed, although I don't know why.

crdelsey, 2019-11-06:
*Update*: This broke when we updated to `behaviortree_cpp_v3`. The cause of this is a change in the way `Sequence` nodes work.

There is no direct replacement for the `Sequence` node in the new library. Possible solutions are:
1. Reimplement the Version2 sequence node as a `LegacySequence`.
2. Make the tree work again using the `Parallel` node in some way.
3. Change the behavior of the upstream `ReactiveSequence` so that it can be used as a drop in replacement.

PR #1371 (2019-11-22): Fix parallel replanning in the behavior tree (alternate implementation)
## Basic Info

| Info | Please fill out this column |
| ------ | ----------- |
| Ticket(s) this addresses   | fixes #1301, fixes #1307 |
| Primary OS tested on | Ubuntu|
| Robotic platform tested on | TB3 in Gazebo |

---

## Description of contribution in a few bullet points

* This is an alternate implementation of #1336 that the team mostly prefers.
* This uses an alternate control node that I'm calling PipelineSequence for lack of a better name. The comments and documentation for this node should explain how it works.  [... cut]
closed 2019-11-22 by crdelsey