ros-navigation/navigation2 #1307: Robot not avoiding obstacles

yathartha3, 2019-10-26:
## Bug report

**Required Info:**

- Operating System:
  - OS and version Ubuntu 18.04
- Version or commit hash:
  - f409ba184e54f919bb8a2cbb6d0057b8b3658dde
- DDS implementation:
  - Fast-RTPS

#### Steps to reproduce issue
Launch navstack.
Send goal pose.
Drop an obstacle on the robot's path.

#### Expected behavior
Calculate new global path and avoids the obstacle.

#### Actual behavior
Obstacle appears in the global costmap, but the path is not updated.
Robot does recovery behavior, and calculates a new path after sometime. 

#### Additional information
[image]

SteveMacenski, 2019-10-26:
I think that’s a direct result of the behavior tree immediately going to recovery mode when anything fails to generate a valid output. If you wrote a quick print statement, it would surprise me if it wasn’t the case that on seeing the obstacle DWB failed, triggered recovery regardless if it just needs a plan update, then replanned.

That long wait I hope is the wait recovery acting... what do your logs say?

yathartha3, 2019-10-26:
@SteveMacenski I updated the issue with the log in the details.

SteveMacenski, 2019-10-26:
Huh failed to make progress really fired fast if that’s the case. Because these logs don’t have stamps, can you tell me if that progress checker claim happened immediately or it happened after dwb sat for a bit and didnt move?

yathartha3, 2019-10-26:
@SteveMacenski The progress checker claim happened after DWB sat for a bit. Even before getting to the recovery behaviors, the global path should be computed every second is what I thought. I did not just drop the obstacle very close to it. It had enough time if a different global path was computed.

SteveMacenski, 2019-10-26:
At least from your gif, I don’t think that’s what’s happening. It looks to me like DWB couldn’t compute a control so it returned the failure and that triggered recoveries. Replanning at 1hz only happens when it wasn’t in a failure part of the tree. 

I’d recommend testing this with just adding a retry-able node around the plan->control with a throttle of 1hz and retry >= 3. You should see that on the second or third retry depending on the async updates of path that a new path is computed and then the control server gets a new path to work with. 

If that works, the long term solution is the retry node that retries probably 3-5ish times before recovery & also adding a blackboard param for ‘force_update_path’ or similar to let DWB tell the planner it needs to update because control failed outside the 1hz update. The state machine in Nav1 did this by switching the state from control to planning on failure.

yathartha3, 2019-10-26:
I put print statements inside ```PlannerServer::computePathToPose()```.
It was not ticking at 1Hz.

orduno, 2019-10-28:
Let's solve #1301 first and then test obstacle avoidance.

yathartha3, 2019-10-28:
> Let's solve #1301 first and then test obstacle avoidance.

I am thinking the same.

PR #1371 (2019-11-22): Fix parallel replanning in the behavior tree (alternate implementation)
## Basic Info

| Info | Please fill out this column |
| ------ | ----------- |
| Ticket(s) this addresses   | fixes #1301, fixes #1307 |
| Primary OS tested on | Ubuntu|
| Robotic platform tested on | TB3 in Gazebo |

---

## Description of contribution in a few bullet points

* This is an alternate implementation of #1336 that the team mostly prefers.
* This uses an alternate control node that I'm calling PipelineSequence for lack of a better name. The comments and documentation for this node should explain how it works.  [... cut]
closed 2019-11-22 by crdelsey