ros-navigation/navigation2 #1309: Add timestamps to all logs

SteveMacenski, 2019-10-28:
Right now its very challenging to debug using the logging to screen or file when you can't tell the relative timing of the events occurring.

this might be something to move upstream to rcl logging but for now I'd like to at least cover Nav2 logging with timestamps.

mjeronimo, 2019-10-29:
Steve, there is a feature in the ROS2 logging subsystem that allows custom formatting of log messages (see ros2/src/ros2/rcutils/src/logging.c). There are several variables that the log system knows about:

- severity
- name
- message
- function_name
- file_name
- time
- time_as_nanoseconds
- line_number

You can use these in the setting of the RCUTILS_CONSOLE_OUTPUT_FORMAT environment variable. For example,

`# export RCUTILS_CONSOLE_OUTPUT_FORMAT="[{time_as_nanoseconds}]: [{severity}] [{name}]: {message}"`

This will include the time, in nanoseconds, that the current message was output.

-Mike

SteveMacenski, 2019-11-06:
Well I think if this exists, we should find a way to apply these settings in the launch files for all of Nav. 

I don't think its optional to include timestamp in logs, that's so critical to debugging async navigation issues, I'm shocked its "batteries-included" but not default.

crdelsey, 2019-11-11:
We can probably add it to the launch files in the same way we set `RCUTILS_CONSOLE_STDOUT_LINE_BUFFERED`

I agree it would be better if this were enabled by default upstream, or at least, if it was much easier to turn on.

SteveMacenski, 2019-11-11:
It's on my backlog to look at, but if someone else is motivated, I certainly wouldn't argue

SteveMacenski, 2019-11-13:
https://github.com/ros2/rcutils/pull/190

SteveMacenski, 2019-11-22:
RCUtils is merged.

closed 2019-11-22 by SteveMacenski